qmk/qmk_configurator #919: [Bug] Lagrange json import/export is buggy

hasufell, 2021-05-08:
I tried both export and import via json2c and c2json and there is some skew with the keymapping. I think it's easiest to describe it with the import:

C code and json exported via `c2json` is here: https://gist.github.com/hasufell/ee9429fcdc7c9f7022125cc7cf1fe36f

When imported, I get this:

[image]

I highlighted the keys that are incorrect in with red circles.

So some keys are swapped with others, some aren't there at all (`ANY(PRN)`) and some shouldn't be there (`KC_LGUI`).

Whether this is a problem with this repo or some other... I don't know. But the keymapping in my C file is correct and works as expected.

noroadsleft, 2021-05-10:
> So some keys are swapped with others, some aren't there at all (`ANY(PRN)`) and some shouldn't be there (`KC_LGUI`).

I've opened a PR to fix most of your issue (linked above this comment), but your `PRN` keycode getting dropped is because [Configurator removes any keycode that is less than 4 characters](https://github.com/qmk/qmk_configurator/blob/3f373e6185227fe5b2bf9680ba4838f294e94c02/src/jquery.js#L419-L428). Only core QMK keycodes are supported in Configurator at this time.

hasufell, 2021-05-10:
awesome, thanks for the quick fix!

closed 2021-06-25 by hasufell