qmk/qmk_configurator #924: [Feature Request] WebUSB Programmer

ecumene, 2021-05-19:
QMK Configurator receives a binary from the API and just provides a download link. It's possible (though experimental, and not in Firefox/Safari) to [flash a new keyboard firmware through the browser](https://devanlai.github.io/webdfu/dfu-util/). This could make it possible for users of things like a Chromebook or Android Phone to flash their keyboard firmware.

WebUSB has a good track record for security and support through Chrome, but other browsers have not expressed any interest in supporting it as it's pretty controversial, and I understand the criticisms too. In my opinion the power it brings to the web is far greater than the risk of "bloat" in some websites. [Here's an example of an Android Operating System you can flash onto a rooted phone via the web right now](https://grapheneos.org/install/web).

This could be an optional feature in the configurator, or if not I plan to implement it into my own site for fun 😄 So how about it? Is this a desirable feature?

yanfali, 2021-05-19:
WebUSB is really cool, but it seems google itself may have moved on to WebHID, as it is in the latest versions of chrome. Have you looked at https://github.com/remap-keys/remap ?

ecumene, 2021-05-19:
No I haven't! That's really neat.

ecumene, 2021-05-19:
Good to know they switched to WebHID.