qmk/qmk_configurator #931: [Bug] Error on compile

chiwou, 2021-05-29:
## Describe the Bug

At the end of the compile I get an error

## Additional Context

actual error text
`
`
Warning: The bin/qmk script is being deprecated. Please install the QMK CLI: python3 -m pip install qmk
Traceback (most recent call last):
  File "bin/qmk", line 57, in <module>
    main()
  File "bin/qmk", line 38, in main
    import qmk.cli  # noqa
  File "/qmk_compiler/qmk_firmware/lib/python/qmk/cli/__init__.py", line 158, in <module>
    from . import console  # noqa
  File "/qmk_compiler/qmk_firmware/lib/python/qmk/cli/console.py", line 9, in <module>
    import hid
ModuleNotFoundError: No module named 'hid'
`

Tried it with different keyboards
massdrop/alt
kbdfans/kbd67
alpine65

always the same

skullydazed, 2021-05-30:
This should be fixed now.

closed 2021-05-30 by skullydazed

chiwou, 2021-05-30:
👍