qmk/qmk_configurator #943: mac os big sur option and cmd confusion [Bug]

gotjoshua, 2021-07-01:
### Describe the Bug

GUI and GUI_T are sending ALT ("option")
ALT alone sends nothing 

I am using one of these:
https://falba.tech/customize-your-keyboard/customize-your-ergodox/

weirdly enough, if i put GUI inside a GUI_T (the one next to space in this ss) i get the inner GUI on tap, but ALT on hold!
[image]
[image]

for the far left key in the ss, holding sends nothing, but tap sends ALT.

### Additional Context?

```json
{
  "version": 1,
  "notes": "",
  "documentation": "\"This file is a QMK Configurator export. You can import this at <https://config.qmk.fm>. It can also be used directly with QMK's source code.\n\nTo setup your QMK environment check out the tutorial: <https://docs.qmk.fm/#/newbs>\n\nYou can convert this file to a keymap.c using this command: `qmk json2c {keymap}`\n\nYou can compile this keymap using this command: `qmk compile {keymap}`\"\n",
  "keyboard": "ergodox_ez",
  "keymap": "ergodox_ez_layout_ergodox_pretty_norman_v22",
  "layout": "LAYOUT_ergodox_pretty",
  "layers": [
    [
      "KC_GRV",
      "KC_1",
      "KC_2",
      "KC_3",
      "KC_4",
      "KC_5",
      "LGUI(KC_TAB)",
      "LGUI(KC_GRV)",
      "KC_6",
      "KC_7",
      "KC_8",
      "KC_9",
      "KC_0",
      "KC_BSPC",
      "KC_TAB",
      "KC_Q",
      "KC_W",
      "KC_D",
      "KC_F",
      "KC_K",
      "LT(2,KC_EQL)",
      "LT(1,KC_QUOT)",
      "KC_J",
      "KC_U",
      "KC_R",
      "KC_L",
      "KC_SCLN",
      "KC_BSLS",
      "LCTL_T(KC_ESC)",
      "KC_A",
      "KC_S",
      "KC_E",
      "KC_T",
      "KC_G",
      "KC_Y",
      "KC_N",
      "KC_I",
      "KC_O",
      "KC_H",
      "KC_ENT",
      "KC_LSFT",
      "KC_Z",
      "KC_X",
      "KC_C",
      "KC_V",
      "KC_B",
      "LSFT_T(KC_SCLN)",
      "RSFT_T(KC_MINS)",
      "KC_P",
      "KC_M",
      "KC_COMM",
      "KC_DOT",
      "KC_SLSH",
      "RSFT_T(KC_UP)",
      "OSM(MOD_HYPR)",
      "OSM(MOD_LSFT|MOD_LALT)",
      "LALT_T(KC_LALT)",
      "RGUI_T(KC_RGUI)",
      "KC_SPC",
      "KC_SPC",
      "KC_APP",
      "KC_LEFT",
      "KC_DOWN",
      "KC_RGHT",
      "OSM(MOD_LSFT|MOD_LALT|MOD_LGUI)",
      "TO(0)",
      "TT(1)",
      "OSM(MOD_LCTL|MOD_LGUI)",
      "KC_HOME",
      "KC_PGUP",
      "LT(2,KC_SPC)",
      "KC_LALT",
      "KC_DEL",
      "LAG_SWP",
      "KC_RALT",
      "LT(1,KC_SPC)"
    ],
    [
      "KC_NO",
      "KC_F1",
      "KC_F2",
      "KC_F3",
      "KC_F4",
      "KC_F5",
      "KC_F6",
      "KC_F7",
      "KC_F8",
      "KC_F9",
      "KC_F10",
      "KC_F11",
      "KC_F12",
      "KC_TRNS",
      "KC_TRNS",
      "KC_EXLM",
      "KC_AT",
      "KC_HASH",
      "KC_DLR",
      "KC_PERC",
      "KC_TRNS",
      "KC_TRNS",
      "KC_CIRC",
      "KC_AMPR",
      "KC_ASTR",
      "KC_LPRN",
      "KC_RPRN",
      "KC_TRNS",
      "KC_TRNS",
      "KC_LT",
      "KC_LCBR",
      "KC_LBRC",
      "KC_LPRN",
      "KC_AMPR",
      "KC_PIPE",
      "KC_RPRN",
      "KC_RBRC",
      "KC_RCBR",
      "KC_GT",
      "KC_TRNS",
      "KC_TRNS",
      "KC_PERC",
      "KC_CIRC",
      "KC_LBRC",
      "KC_RBRC",
      "KC_QUOT",
      "KC_TRNS",
      "KC_TRNS",
      "KC_DQUO",
      "KC_1",
      "KC_2",
      "KC_3",
      "KC_UP",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_DOT",
      "KC_0",
      "KC_EQL",
      "KC_TRNS",
      "RGB_MOD",
      "KC_TRNS",
      "KC_TRNS",
      "KC_NO",
      "KC_TRNS",
      "KC_TRNS",
      "RGB_VAD",
      "RGB_VAI",
      "KC_TRNS",
      "KC_TRNS",
      "RGB_HUD",
      "RGB_HUI"
    ],
    [
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "LALT(KC_UP)",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_PGUP",
      "LALT(KC_LEFT)",
      "KC_UP",
      "LALT(KC_RGHT)",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_1",
      "KC_2",
      "KC_3",
      "KC_4",
      "KC_5",
      "KC_PGDN",
      "KC_LEFT",
      "KC_DOWN",
      "KC_RGHT",
      "KC_TRNS",
      "KC_MPLY",
      "KC_TRNS",
      "KC_6",
      "KC_7",
      "KC_8",
      "KC_9",
      "KC_0",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "LALT(KC_DOWN)",
      "KC_MNXT",
      "KC_UP",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_BTN1",
      "KC_BTN2",
      "KC_VOLU",
      "KC_VOLD",
      "KC_LEFT",
      "KC_DOWN",
      "KC_RGHT",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_WBAK"
    ],
    [],
    [],
    [],
    [],
    [],
    [],
    [],
    [],
    [],
    [
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_TRNS",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO"
    ]
  ],
  "author": ""
}
```

gotjoshua, 2021-07-01:
partial workaround:

[image]

but i have no way to use ALT_T nor GUI_T

yanfali, 2021-07-01:
That's very odd behavior. Do you have any other keyboards running QMK that have the same issue?

yanfali, 2021-07-01:
Also, does it do with this Right Alt and Right Gui keycodes? or Only Left.

drashna, 2021-07-01:
Have you tried clearing EEPROM on the keyboard?

gotjoshua, 2021-07-01:
> That's very odd behavior.

I do totally agree!
I am running a hackintosh (so maybe its something with that)

> Right Alt and Right Gui

unfortunately both.

I have a bluetooth keyboard (not running QMK) that works fine, and is usually off.

i uninstalled Karabiner and USBoverdrive (to be sure they were not interfering)

also just noticed that `'~` is sending ESC, but also works correctly if wrapped in a modifier like `LGUI(KC_GRV)`

i have 2 things i'd like to try
1. Catalina
2. different locales and keyboard types

gotjoshua, 2021-07-01:
> Have you tried clearing EEPROM on the keyboard?

well i clicked that button on qmk toolkit, but got no feedback that it did anything... is there another way to do that?

also re-flashed many times in troubleshooting mode

yanfali, 2021-07-01:
Because you're using configurator, I think adding a RESET keycode to your keymap is the simplest option. Add the keycode to a layer then reflash.

Once your keyboard is running, press RESET and then unplug and replug the keyboard and your eeprom should be reset.

This kind of problem is one of the reasons we don't promote bootmagic because it causes footguns.

gotjoshua, 2021-07-01:
> we don't promote bootmagic because it causes footguns

i'm afraid that's a bit over my head... can you translate?

gotjoshua, 2021-07-01:
> press RESET and then unplug and replug the keyboard and your eeprom should be reset.

tried this twice, but same:

-  pure ALT presses don't get to the OS
- ~ sends ESC
- GUI only works when wrapped (like i show in the workaround above)

yanfali, 2021-07-01:
KC_GRV is broken on mac. I think this is documented, it's because it
conflicts with a built in key binding on the MacOS. Not sure what's going
on with your keyboard. You could also check your modifier settings in the
keyboard control panel.

gotjoshua, 2021-07-02:
> Not sure what's going on with your keyboard. 
Me neither! is there a good way to debug this stuff closer to the bits... i mean how can i see what happens in the arduino controller when i press ALT and nothing comes through to the Karabiner-EventViewer ?

> You could also check your modifier settings in the keyboard control panel.
this i did and tried some work arounds, but nothing helps as the pure ALT does nothing : (

I assume either it is something with recent changes to the compiler and/or something weird with my big sur hackintosh.

Let's see if anyone else starts reporting issues or finding this one... 
If not, then I guess I'll need to settle for my workarounds.

Many Many thanks for your time and support!

yanfali, 2021-07-10:
Since we're not able to duplicate this on any machine except the reporter, I may close this one.

closed 2021-08-28 by yanfali