qmk/qmk_configurator #954: [Bug] Some keyboard layouts bug to just a single line

AMoonRabbit, 2021-07-14:
### Describe the Bug

When selecting some of the templates available for a keyboard, the keys all move to just a single line

This has only been found with the coseyfannitutti/discipline keyboard layouts, specifically when selecting LAYOUT_65_ISO, note that if the ISO block version is selected, it loads fine.

[image]

### Additional Context?

[Discipline keyboard layout](https://config.qmk.fm/#/coseyfannitutti/discipline/LAYOUT_65_ISO)

AMoonRabbit, 2021-07-14:
Could be the same issue as #902?

yanfali, 2021-07-15:
> Could be the same issue as #902?

Yes. There is no LAYOUT_65_iso layout defined in the layouts in info.json. It's being inferred by the header files.

yanfali, 2021-07-15:
@noroadsleft this is really a QMK firmware bug I believe.

noroadsleft, 2021-07-16:
> @noroadsleft this is really a QMK firmware bug I believe.

Yeah, this has to be fixed on `qmk_firmware`. Configurator's looking for data that doesn't exist.

I can sort this.

noroadsleft, 2021-07-18:
This is fixed for the Discipline. Any layout you select for it should be fully functional.

If you spot any bugs similar to this for any other boards in the future, posting an issue is greatly helpful. I don't have a good method for checking for these sorts of issues at the moment.

Thanks for reporting!

closed 2021-07-18 by noroadsleft

AMoonRabbit, 2021-07-18:
Thank you for fixing this