qmk/qmk_configurator #955: One-shot shift not working in combination with hold layers[Bug]

jweickm, 2021-07-17:
### Describe the Bug

Hi QMK community,
The OSM LShift function works when it is the only function mapped to a key. 
However when I want to use it in combination with some "kc on tap switch to layer xyz while held" it does not work (tested on my olkb planck/rev6. See attached image. 
Would be grateful for help. 
Jakob

### Additional Context?

[image]

noroadsleft, 2021-07-18:
I think I may know what's going on here, but your linked image isn't actually linked, so I can't be sure.

jweickm, 2021-07-18:
> I think I may know what's going on here, but your linked image isn't actually linked, so I can't be sure.

Thanks for the reply. Something must have gone wrong when uploading the picture. It is just an image of how I tried to implement this feature on one of my keys. 
As OSM LSft also has a hold feature, does it not work together with other "kc on tap switch to layer xyz while held" modifiers?

noroadsleft, 2021-07-18:
Yep, it's exactly what I suspected.

The inner argument for [Layer Tap keys](https://docs.qmk.fm/#/feature_layers?id=switching-and-toggling-layers) is limited to the [Basic Keycode set](https://docs.qmk.fm/#/keycodes_basic) - if it's not on this page, you can't use it inside an `LT()` keycode.

[The QMK Docs detail this](https://docs.qmk.fm/#/feature_layers?id=caveats), screenshot below.

[image]

jweickm, 2021-07-18:
Hi James,
thanks for clarifying this. Would this be something that can be implemented with QMK, just not the online configurator?
j

On 2021年7月18日 17:23:18 GMT+02:00, James Young ***@***.***> wrote:
> Yep, it's exactly what I suspected.
>
> The inner argument for [Layer Tap
> keys](https://docs.qmk.fm/#/feature_layers?id=switching-and-toggling-layers)
> is limited to the [Basic Keycode [...]

Jakob Weickmann

yanfali, 2021-08-28:
Since this is an advanced use case and not something the configurator currently can support, closing.

closed 2021-08-28 by yanfali