qmk/qmk_configurator #956: Reconfiguring Pins[Feature Request]

lvnvs, 2021-07-19:
### Description

I have a snagpad with a faulty F7 pin, so I just handwired the column to another pin. However I can only use KBfirmware to change the pin, which currently works, but I still want to use QMK Configurator to set up the keymap.

noroadsleft, 2021-08-02:
This isn't feasible to do at this point in time. QMK Configurator doesn't actually compile keymaps, it only generates them and sends them to another service which does the compilation. There's no interface in that system to redefine the pins that are used by a board.

If you have a local Git clone of the qmk/qmk_firmware repository, you can import your Configurator `keymap.json` to that and change it for only your keymap, and compile it in QMK Firmware instead of Configurator.

yanfali, 2021-08-28:
Closing as question was answered.

closed 2021-08-28 by yanfali