qmk/qmk_configurator #957: [Feature Request] 180 degrees rotation for grid keyboards

JhorAVi, 2021-07-23:
### Description

Most grid keyboards like planck and preonic have USB cables located on top which is a big obstacle when working with laptops resulting you to push the screen farther away. Rotating the keyboard 180 degrees clears the wire obstacle allowing the keyboard to touch the laptop for more stable support. I have successfully rotated the keyboard by editing the layout map and typing experience is much better actually. A rotation define statement will allow this feature much quicker.

yanfali, 2021-08-02:
https://www.amazon.com/UGREEN-Charging-Compatible-Samsung-Nintendo/dp/B081SFHFTF/ref=sr_1_4?dchild=1&keywords=USB+C+RIGHT+ANGLE+CABLE&qid=1627928900&sr=8-4

yanfali, 2021-08-02:
Not all problems need a solution in software

closed 2021-08-28 by yanfali