qmk/qmk_configurator #962: Take over some of the features from Ruiqi Mao

TorstenC, 2021-07-31:
### Features from kbfirmware.com

In the unofficial Keyboard Firmware Builder from Ruiqi Mao https://kbfirmware.com/ are additional features available.

Did you already discuss to take over some of these features to config.qmk.fm or to merge these projects?

If the discussion is happening [in Discord](https://discord.com/channels/440868230475677696/867530744116543508/870972960461164544), here just to record the outcome and avoid duplicate questions.

PS: Ruiqi Mao writes about his Firmware Builder: _"It is strongly encouraged to migrate any projects to more updated solutions."_
His repo: https://github.com/ruiqimao/qmkbuilder

noroadsleft, 2021-07-31:
Which specific features were you looking for?

We have some ongoing discussions about features/functionality that would be good to have in QMK Configurator, but nothing is set in stone at the moment.

TorstenC, 2021-07-31:
Thanks for the comments in the [channel #qmk_configurator](https://discord.com/channels/440868230475677696/867530744116543508).

Merging these projects is not an option, because KLE JSON doesn't contain nearly enough information to generate firmware out of it and that's not the way, configurator currently works.

> Which specific features were you looking for?

Thanks for your interest, @noroadsleft.

**I am looking for a direct import from KLE JSON to qmk_configurator JSON.** 

_My understanding: KLE JSON is JSON from keyboard-layout-editor.com and I assume, "qmk_configurator JSON" is also called "info.json"._

For me, it's mostly about avoiding iterations in development:

A direct import would help to think more carefully about the assignment of the keys in the layers for a newly invented keyboard layout, before designing new schematics and the new PCB layout. Hardware, firmware and contributing the result to the qmk_firmware repo would be done later, after the new concept is ready and the PCBs are ordered e.g. from JLCPCB.
This seems to be a new approach that would avoid unnecessary development iterations and increase development speed, I imagine.

noroadsleft, 2021-07-31:
> KLE JSON doesn't contain nearly enough information to generate firmware out of it

KLE is a separate project from kbfirmware.com, and neither project is one over which QMK has control. Merging KLE and QMK Configurator doesn't make sense, because KLE is about designing keysets, and QMK Configurator is about programming keyboards. Tools like kbfirmware.com can be made to accept KLE JSON, and in fact QMK has some tooling that converts KLE JSON into `info.json` data, but the purposes for each site and set of tooling is different.

> A direct import would help to think more carefully about the assignment of the keys in the layers for a newly invented keyboard layout, before designing new schematics and the new PCB layout.

We've talked internally in the past about giving QMK Configurator support for random handwired boards (which is something for which kbfirmware.com is well-suited). A data-driven QMK could support this use case, but the API that supports Configurator would still need to be able to understand a "full keyboard source" submission like, and that is a major task.

yanfali, 2021-08-28:
This isn't really a question, I'm going to close it.

closed 2021-08-28 by yanfali