qmk/qmk_configurator #964: "Any" blocks are ignored on import

hjklapp, 2021-08-01:
### Describe the Bug

When I create a keymap using the "Any"-key to specify custom keycodes, they are exported to json.

When I import the same json file into the configurator, the "Any"-keys are not imported. Instead, I see the triangle-key (fall through).

My expectation is that the "Any"-keys are imported and displayed again as "Any"-keys, including the specified custom keycode. Otherwise working with custom keys in the configurator is very cumbersome.

noroadsleft, 2021-08-01:
Could you post your `keymap.json` file? I just went through a bunch of keymap updates last week and the `ANY()` functionality didn't give me any issues.

hjklapp, 2021-08-01:
Sure.
The problem are custom keycodes like `OSS` which are not part of qmk.
So just including `ANY(OSS)` in a keymap should trigger the problem.

[hjklapp.json.log (remove `.log`)](https://github.com/qmk/qmk_configurator/files/6913375/hjklapp.json.log)

noroadsleft, 2021-08-02:
K, you have two problems here:

1. The `ANY()` keycode is for meant for keycodes that are not understood by QMK Configurator, but are defined either in QMK's core or in the keyboard's own config (some keyboards define extra keycodes that are available to any keymap for that keyboard with no action required on the part of the user). As your keymap is for a Kyria, if the Kyria doesn't define the keycodes you're trying to use, and they also aren't part of QMK, the compile attempt will fail because the keycode does not exist.

2. As none of the keycodes in QMK are shorter than four characters, QMK Configurator removes any keycodes that are shorter than four characters long from the keymap on import, and replaces them with `KC_TRNS` (the "fall-through" keycode).
   https://github.com/qmk/qmk_configurator/blob/f035bb39648068761e73dcf52ca1c98fc5925f55/src/jquery.js#L419-L428

hjklapp, 2021-08-02:
1. I don't use the configurator to compile. I use it as a graphical tool to edit my keymap. The custom keycodes are defined in my code, so no compile problems for me :)

2. Ah, thank you! Also for the pointer to the code part. I was playing around yesterday and found that my longer custom keycodes are not removed. So an immediate solution to my problem would be to make my custom keycodes longer. 

But ... how useful is that restriction? I mean, the configurator still can't compile my keymap.
And I really like my short custom keycode names (eg UE is a very good description of the umlaut Ü).

If I were to file a feature request to remove that check, would it have a chance?

If the check is meant to stay, it would be nice to also add it to the GUI. Eg throw a warning when ANY is used with less than 4 characters of custom code. 

What do you think about that?

yanfali, 2021-08-02:
We specifically ignore User defined macros in keymaps because we don't have a good way to figure them out on the configurator at this time, as it's a compile time C preprocessor expansion. There's hundreds of keyboards and thousands of keymaps in QMK so it's not particularly practical for us. This may change when data driven keymaps is full working. There should be an error message in the fake console when you import the keymap, and the error message you should see is in the code above.

yanfali, 2021-08-02:
Looks like the error message is broken. So that's a bug.

yanfali, 2021-08-02:
Ah, ok there's a race condition while it's loading up the new keyboard where it overwrites the error message.

yanfali, 2021-08-02:
[image]

This should at least be surfaced better now, so you should get a visual warning that it's happening.

yanfali, 2021-08-02:
Referenced in commit 40bd4224c28a3030cf8e8fcd7b6f111625b1ad35

hjklapp, 2021-08-02:
I can confirm that I see the error messages on import.

> We specifically ignore User defined macros in keymaps because we don't have a good way to figure them out on the configurator at this time

I do not understand this. Just display them as `ANY(customKeycode)`.
Maybe along with a warning in the console.
But I probably don't get the bigger picture here.

I think it is a decision whether the configurator should be a "stand alone tool" that can build on the web (that's how I used it to get started), or also for people that compile on their own machines and use it in conjunction with custom code. I think it could be both.
And if I use custom keycodes longer than 3 characters, it already is :)

Thank you very much for your help and for building the awesome configurator!

closed 2021-08-02 by hjklapp

yanfali, 2021-08-02:
> I can confirm that I see the error messages on import.
>
>
> I do not understand this. Just display them as `ANY(customKeycode)`.
> Maybe along with a warning in the console.
> But I probably don't get the [...]

A bad user experience is one where we pass through unknown keycodes to a backend that may or not be able to deal with them. CPP (C preprocessor) is basically a infinite possibilities language. Allow anything to be executed would potentially be a security or denial of service issue; think about why docker/github have now banned running arbitrary PRs on most projects because people have been using bitcoin mining. To avoid issues with difficult to explain errors, we choose to only support known QMK keycodes, not specials, aliases or ones that are convenient. I'm sorry if this goal conflicts with your personal preferences, but that's just the way it is at this time.

> I think it is a decision whether the configurator should be a "stand alone tool" that can build on the web (that's how I used it to get started), or also for people that compile on their own [...]

These are all great ideas, but I would remind you Configurator isn't a stand alone tool though. It's a front end of the API maintained by @skullydazed .  Without the API the configurator is a useless piece of eye candy. Anyone is encouraged and permitted to use the API directly and and build in any abstractions they would prefer.

hjklapp, 2021-08-03:
@yanfali Thank you for the explanation!