qmk/qmk_configurator #967: [Bug]

ryanaungs, 2021-08-09:
### Describe the Bug

When loading https://config.qmk.fm/#/ there is no option to let me select my keyboard. Not sure if this is a bug, or I am doing something wrong. 
- I do not have the QMK keyboard plugged into my MAC. Is this a prerequisite to selecting the keyboard? 
- I have installed toolbox on my MAC.
Here is a screenshot of what I see.
[image]

Any ideas?

noroadsleft, 2021-08-09:
Known issue. The task that provides the list of keyboards to QMK Configurator crashed during an update Sunday evening, and it wasn't discovered until after any of the people who are able to fix it signed off for the night.

I imagine this will be fixed sometime early Monday (US Pacific Time, UTC -0700).

skullydazed, 2021-08-09:
The problem has been fixed by https://github.com/qmk/qmk_firmware/pull/13930.

closed 2021-08-09 by skullydazed

xtaran, 2021-10-09:
Maybe this issue should be unpinned as it seems solved for about two months now. :-)