qmk/qmk_configurator #985: [Bug] Just the equal sign does not work when flashing to a iris, have to use the + and = on shift key to use equal sign

kleenkanteen, 2021-08-30:
### Describe the Bug

My keyboard layout's JSON: https://mega.nz/file/fbAhAQ7J#pxgC5Bd5aEaygkpj0HVAioxXf-YBAp_WbYFY9GWw_2A

Compiled it 3 times, flashed 3 times, equal sign still does not work on the first layer. I have to use the key that is = by default and + on shift. Happened to me before actually as well, encountered again when changing my layout.

sigprof, 2021-08-30:
The key with just `=` is `KC_PEQL` — the “Keypad =” key, which works only in Mac OS (because Apple keyboards actually have that key); in Windows a press of that key is apparently translated to some key event, but it's not mapped to any character. So this is not really a bug in QMK — the HID “Keyboard/Keypad Page” contains lots of keycodes, but anything which is not present on a standard 104/105-key keyboard may not really be supported by the software on the computer side.

kleenkanteen, 2021-08-30:
Hmm, interesting. Thanks, maybe in the future there could be a way to notify windows users about this.

closed 2021-08-30 by kleenkanteen