qmk/qmk_configurator #986: [Bug] GMMK Pro lacking the function layer on the online configurator

andrebrait, 2021-08-31:
## Describe the Bug

The GMMK Pro is lacking the function layer on the online configurator. I've checked other keyboards that were added recently and they all show up just fine. I couldn't figure out why the GMMK Pro is only showing the keys assigned on the first layer.

I checked a number of information files and whatnot but everything seems to be the same for other keyboards that show up correctly (like the Keychron Q1).

## System Information

 - Keyboard:  GMMK Pro
   - Revision (if applicable):  
 - Operating system: Irrelevant
 - AVR GCC version: Irrelevant

 - ARM GCC version: Irrelevant

 - QMK Firmware version: Irrelevant

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

andrebrait, 2021-08-31:
Can you point me to where I can fix it, in this case?

fauxpark, 2021-08-31:
Configurator keymaps are stored here:
https://github.com/qmk/qmk_configurator/tree/master/public/keymaps

yanfali, 2021-08-31:
https://github.com/qmk/qmk_configurator/blob/master/public/keymaps/g/gmmk_pro_ansi_default.json
https://github.com/qmk/qmk_configurator/blob/master/public/keymaps/g/gmmk_pro_iso_default.json

andrebrait, 2021-08-31:
Ah, perfect.

Thanks.

I shall make a PR for those then.

Andre Brait

andrebrait, 2021-08-31:
Done.

PR #987 (2021-08-31): [BUG] Sync layer 1 of gmmk/pro/ansi with qmk_firmware

## Description

Fixed #986
The ISO layout was already working correctly.
commit 2f85a740 (2021-08-31): Sync layer 1 of gmmk/pro/ansi with qmk_firmware (#987)

Closes #986
closed 2021-08-31 by noroadsleft