qmk/qmk_configurator #993: JSON files with pipe symbols in Any key blocks fail to import correctly

archy3, 2021-09-07:
### Describe the Bug

I use the `Any` key and place `OSM(MOD_LSFT|MOD_RALT)` in key to the right of `OSM RAlt` on the bottom row of layer zero of the attached layout (this is useful if RAlt is used at AltGr) (rename the extension from "log" to "json").

If I compile this, it compiles fine and flashing the firmware works as expected.
The bug is if I export to a JSON file and import it again, layer zero will be blank. Experimentation has led me to believe any `Any` block with a pipe symbol will result in the same problem.

### Additional Context?

[planck_rev6_layout_ortho_4x12.log](https://github.com/qmk/qmk_configurator/files/7118138/planck_rev6_layout_ortho_4x12.log)

yanfali, 2021-10-22:
I'm going to guess this is an import issue. It's not breaking up the imported keycode into smaller values. So while it works when input the first time, it's not restored.

yanfali, 2021-10-22:
There's a bug. When both a left and a right mod are used, the right mod overrides the left mod. So what's getting sent is RSFT|RALT which isn't what was input.

archy3, 2021-10-22:
For my particular use case, `OSM(MOD_RSFT|MOD_RALT)` would work just as well as `OSM(MOD_LSFT|MOD_RALT)`.