qmk/qmk_configurator #995: [Bug] Lily58/rev1 - Online Configurator Compile hex file is smaller than default hex file - Incomplete compile

moonnoonn, 2021-09-23:
### Describe the Bug

Online Configurator: Using the firmware from this source my keypresses only register on the left board and not the right board. I tried with my custom layout and the default layout, both files resulted the same issue. Both hex file size was 52kb

QMK Tool - Use the default firmware and both left and right board register all key presses. The hex file was 59kb.

I tried with a different keyboard, there were no issues flashing the firmware that was downloaded from the online Configurator.

This issue is only affecting Lily58/rev1 and Lily58/light keyboard

Log from flashing with QMK Tool:
*** Atmel DFU device connected: ATMEL ATm32U4DFU (03EB:2FF4:0000)
*** Attempting to flash, please don't remove device
>>> dfu-programmer atmega32u4 erase --force
    Erasing flash...  Success
    Checking memory from 0x0 to 0x6FFF...  Empty.
>>> dfu-programmer atmega32u4 flash --force /Users/thu/Downloads/lily58_rev1_default_10F0ECA5.hex
    0%                            100%  Programming 0x5200 bytes...
    [>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
    0%                            100%  Reading 0x7000 bytes...
    [>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
    Validating...  Success
    0x5200 bytes written into 0x7000 bytes memory (73.21%).
>>> dfu-programmer atmega32u4 reset
*** Atmel DFU device disconnected: ATMEL ATm32U4DFU (03EB:2FF4:0000)

### Additional Context?

# Lily58

Lily58 is 6×4+5keys column-staggered split keyboard.

[image]
[image]
Keyboard Maintainer: [Naoki Katahira](https://github.com/kata0510/) [Twitter:@F_YUUCHI](https://twitter.com/F_YUUCHI)  
Hardware Supported: Lily58 PCB, ProMicro  
Hardware Availability: [PCB &amp; Case Data](https://github.com/kata0510/Lily58)

Make example for this keyboard (after setting up your build environment):

    make lily58:default

See the [build environment setup](https://docs.qmk.fm/#/getting_started_build_tools) and the [make instructions](https://docs.qmk.fm/#/getting_started_make_guide) for more information. Brand new to QMK? Start with our [Complete Newbs Guide](https://docs.qmk.fm/#/newbs).

Loaded 4 layers and 232 keycodes. Defined 0 Any key keycodes
* Sending lily58/rev1:lily58_rev1_layout_mine with LAYOUT
* Received job_id: f24bd391-579b-4c30-87e4-31b9a44529df
* Running . . . . . . . .
* Finished:
Ψ Compiling keymap with make -s --jobs=1 -r -R -f build_keyboard.mk KEYBOARD=lily58/rev1 KEYMAP=lily58_rev1_layout_mine KEYBOARD_FILESAFE=lily58_rev1 TARGET=lily58_rev1_lily58_rev1_layout_mine KEYBOARD_OUTPUT=.build/obj_lily58_rev1 KEYMAP_OUTPUT=.build/obj_lily58_rev1_lily58_rev1_layout_mine MAIN_KEYMAP_PATH_1=.build/obj_lily58_rev1_lily58_rev1_layout_mine MAIN_KEYMAP_PATH_2=.build/obj_lily58_rev1_lily58_rev1_layout_mine MAIN_KEYMAP_PATH_3=.build/obj_lily58_rev1_lily58_rev1_layout_mine MAIN_KEYMAP_PATH_4=.build/obj_lily58_rev1_lily58_rev1_layout_mine MAIN_KEYMAP_PATH_5=.build/obj_lily58_rev1_lily58_rev1_layout_mine KEYMAP_C=.build/obj_lily58_rev1_lily58_rev1_layout_mine/src/keymap.c KEYMAP_PATH=.build/obj_lily58_rev1_lily58_rev1_layout_mine/src VERBOSE=false COLOR=true SILENT=false QMK_BIN=qmk

avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/lily58/lily58.c                                                                
Compiling: keyboards/lily58/rev1/rev1.c                                                             
Compiling: .build/obj_lily58_rev1_lily58_rev1_layout_mine/src/keymap.c                              
Compiling: quantum/quantum.c                                                                        
Compiling: quantum/send_string.c                                                                    
Compiling: quantum/bitwise.c                                                                        
Compiling: quantum/led.c                                                                            
Compiling: quantum/action.c                                                                         
Compiling: quantum/action_layer.c                                                                   
Compiling: quantum/action_macro.c                                                                   
Compiling: quantum/action_tapping.c                                                                 
Compiling: quantum/action_util.c                                                                    
Compiling: quantum/eeconfig.c                                                                       
Compiling: quantum/keyboard.c                                                                       
[... 112 lines cut ...]
Compiling: tmk_core/common/avr/bootloader.c                                                         
Assembling: tmk_core/common/avr/xprintf.S                                                           
Compiling: tmk_core/common/avr/printf.c                                                             
Compiling: tmk_core/protocol/lufa/lufa.c                                                            
Compiling: tmk_core/protocol/usb_descriptor.c                                                       
Compiling: lib/lufa/LUFA/Drivers/USB/Class/Common/HIDParser.c                                       
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Device_AVR8.c                                        
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/EndpointStream_AVR8.c                                
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Endpoint_AVR8.c                                      
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Host_AVR8.c                                          
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/PipeStream_AVR8.c                                    
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Pipe_AVR8.c                                          
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.c                                 
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBInterrupt_AVR8.c                                  
Compiling: lib/lufa/LUFA/Drivers/USB/Core/ConfigDescriptors.c                                       
Compiling: lib/lufa/LUFA/Drivers/USB/Core/DeviceStandardReq.c                                       
Compiling: lib/lufa/LUFA/Drivers/USB/Core/Events.c                                                  
Compiling: lib/lufa/LUFA/Drivers/USB/Core/HostStandardReq.c                                         
Compiling: lib/lufa/LUFA/Drivers/USB/Core/USBTask.c                                                 
Compiling: tmk_core/protocol/lufa/usb_util.c                                                        
Linking: .build/lily58_rev1_layout_mine.elf                                                         
Creating load file for flashing: .build/lily58_rev1_layout_mine.hex                                 
Copying lily58_rev1_layout_mine.hex to qmk_firmware folder                                          
Checking file size of lily58_rev1_layout_mine.hex                                                   
 * The firmware size is fine - 18452/28672 (64%, 10220 bytes free)

sigprof, 2021-09-23:
Did you reflash both halves with the new firmware?

The difference in the firmware size is probably because the `default` keymap for Lily58 includes some OLED code, but the configurator does not use that keymap, and that code is not included in configurator builds.

moonnoonn, 2021-09-23:
Yes. I flashed the left first and then flash the right. After that, I move the USB back to the left because it's the default. The right side won't register any key presses unless the USB is plugged to the right. I tried a previous custom .hex file  (retrieve from the configurator) last month and both sides work after flashing them. This is the 2nd Lily58 board that I help build. I'm having a puzzle moment since the 1st Lily58 board did not show any of these issues. I may have to review the hardware and try again to see where the fault is. Next step is to use my built environment instead of using the online configurator firmware file and flash from there. I appreciate your help very much.

moonnoonn, 2021-09-24:
I flashed both halves again with the configurator hex file and the Right board does not register any keys. I clear the EEPROM on both board, flash the configurator hex file, and the right board still having the same problem. After that, flashing the default hex file on both board register all my keypresses. My other Lily58 board was built a month ago and did not have this issue. I'm hesitant to perform the flash if this second set is having issues.

robkuan, 2023-04-06:
I'm having the same exact issue – wondering if anybody found a resolution to this? 

I had a functional Lily58 keyboard, but after flashing using the QMK configurator hex file separately on both halves, my right side keyboard also won't register unless the USB cable is plugged into the right. 

Would appreciate any guidance on troubleshooting!

moonnoonn, 2023-04-06:
Hi my dood,

Try finding the Lily58 default original hex file then flash it to see if the slave side will register with the master. It's an indicator that possibly the hex file you created may have some error in the coding, but it wasn't detected during the flashing. Let me know how it goes for you.

keyboard-magpie, 2023-04-06:
@robkuan nothing to do with configurator- more likely down to the hardware you are using- you'll need to compile your own firmware, adding `#define SPLIT_USB_DETECT` and `#define SPLIT_WATCHDOG_ENABLE` to your keymap level config.h file.

robkuan, 2023-04-07:
thanks all –

@moonnoonn I flashed the default original hex file, and it fixed the error! 

I created a new hex file using the QMK configurator, and everything seemed to be working again. 

@keyboard-magpie I'm not sure what happened the first time around, but you're right in that it doesn't seem to be a configurator problem.