qmk/qmk_configurator #996: [Bug] kinesis/kint36 firmware download for flashing as .hex instead of .bin

negatv, 2021-09-28:
### Describe the Bug

The firmware download compiled for the kinesis/kint36 is .bin format instead of .hex required for flashing on the Teensy 3.6.  

The .hex file is created, but is not available to download.   I took a look over the API documentation to see if there was a method to retrieve the .hex via the API or an option to specify the firmware format generated to no avail.

Building from QMK works fine, as I'm able to retrieve the .hex directly from .build/.  But for those using the configurator, do not have that option.

```
Creating binary load file for flashing: .build/kinesis_kint36_qmkconfigt36base.bin                  
Creating load file for flashing: .build/kinesis_kint36_qmkconfigt36base.hex                         

Size after:
   text	   data	    bss	    dec	    hex	filename
      0	  21412	      0	  21412	   53a4	.build/kinesis_kint36_qmkconfigt36base.hex

Copying kinesis_kint36_qmkconfigt36base.bin to qmk_firmware folder              
```                    

negatv, 2021-10-01:
I have found that adding the following to rules.mk (/keyboards/kinesis/kint36/rules.mk) makes it to copy the .hex instead of bin to the qmk_firmware folder:

`FIRMWARE_FORMAT = hex`

I have not dug into it further, but I suspect that based on the MCU type that this is either being set to .bin, either by default or done so on purpose.

tzarc, 2021-10-01:
Virtually all ARM-based boards use DFU for uploads, and that requires a BIN file.
The teensy's using Halfkay require hex. I imagine this is just an accidental omission in qmk_firmware from the original PR submitter -- your change to `FIRMWARE_FORMAT = hex` is the right way forward.

The one downside is that there is no new release of the `teency_loader_cli` which other parts of the QMK ecosystem rely on for flashing, namely QMK Toolbox. You need a custom build of the current git repository of `teency_loader_cli` to even flash the new teensy's, and we can't provide those binaries until a new release is cut.

A new release been requested on their repo, but as yet has not been actioned.

negatv, 2021-10-01:
Teensy loader works for flashing the 3.6 and 4.1 boards in the meantime, but by changing the default firmware_format to hex would allow those using the configurator to be able to download the correct file for flashing.

I'll try submitting a PR for these small changes to qmk_firmware.

negatv, 2021-10-01:
The PR that I submitted was merged into master.  Tested compiling new firmware for kint36 and is now working as expected.

closed 2021-10-01 by negatv