qmk/qmk_configurator #997: [Bug] kinesis/kint41 firmware does not successfully compile

negatv, 2021-09-28:
### Describe the Bug

Firmware for kinesis/kint41 does not currently successfully compile using the configurator.   

I have pulled latest code from master and was able to build locally.

```
Compiling: keyboards/kinesis/kinesis.c                                                             In file included from ./lib/chibios/os/common/ports/ARMCMx/chcore.h:70:0,
                 from ./lib/chibios/os/rt/include/ch.h:112,
                 from ./lib/chibios/os/hal/osal/rt-nil/osal.h:32,
                 from ./lib/chibios/os/hal/include/hal.h:28,
                 from tmk_core/common/chibios/platform_deps.h:18,
                 from quantum/quantum.h:18,
                 from keyboards/kinesis/kint41/kint41.h:19,
                 from keyboards/kinesis/kinesis.h:20,
                 from keyboards/kinesis/kinesis.c:1:
./lib/chibios-contrib/os/common/startup/ARMCMx/devices/MIMXRT1062/cmparams.h:70:24: fatal error: MIMXRT1062.h: No such file or directory
 #include "MIMXRT1062.h"
                        ^
compilation terminated.
 
make: *** kconfig41/keyboards/kinesis/kinesis.o] Error 1
```

noroadsleft, 2021-09-28:
I've found the cause of this error, but I need to check with someone about a proposed fix – I'm not familiar with the infrastructure where this needs to be fixed.

stapelberg, 2022-02-23:
Can you elaborate on what the new problems are? Maybe someone has a good idea on how to fix it :)

negatv, 2022-02-23:
It's been a while, but when initially building from source for the Teensy 4.1, I had to 'make git-submodule' and it resolved the issue.   Can't say that is the solution in this case, but the error message does point to not being able to find the header file "MIMXRT1062.h".

fauxpark, 2022-02-23:
This issue has to do with the mcux-sdk submodule inside of chibios-contrib. The QMK API tries to determine which submodules to pull down before compiling a board, but it does not do it recursively (and shouldn't). The options here are to recurse checking out submodules for *only* chibios-contrib, or to add custom code for mcux-sdk. Both are not great solutions.

@stapelberg what was the reason for adding this, and is there a way to eliminate the need for it?