qmk/qmk_configurator #1005: [Feature Request] Improved keyboard navigation

thekingofspain, 2021-10-10:
### Description

Its quite hard to quickly navigate through the all the options in the main input select/autocomplete html controls used in the config.qmk.fm page:

1. After a selection, the focus on the html control is not preserved; e.t. _Keyboad_ input html control.
2. If focus is preserved after a selection on the html control, there is no way to change the selected option via the standard keyboard web methods: e.t. _Keymap_ input html control.

Standard html input select controls have the following functionality:

- After selected value changes, the html control remains in foucus .
- The _Enter_ key is used to open and close the select options list.
- The _left_ and _right_ arrow keys are used to change the selected option when the select option list is in a closed state.
- The _up_ and _down_ arrow keys are used to navigate the select options when the select option list is open. 

see https://www.w3schools.com/tags/tryit.asp?filename=tryhtml5_select_form for a plain example of the html control and its navigation.

yanfali, 2021-10-10:
Please add screenshots. We have some non standard drop downs that support search.

thekingofspain, 2021-10-17:
I added some CSS to page for these screen shots:

:focus {background: lightblue !important;}

KEYBOARD - looses focus after selection and closing the selection via the ESC key.

[image]

[image]

LAYOUT - Keeps focus after selection and closing of an open selection list via the ESC key.  Does not respond to the normal keyboard commands to open the selection list and changing the selected item while the select list is closed.

[image]

[image]

KEYMAP:  Keeps focus after selection and the closing the selection list via the ESC key.   Does not respond to the normal keyboard commands to open the selection list and changing the selected item while the select list is closed.

[image]

[image]

yanfali, 2021-10-17:
Feel free to open a PR