qmk/qmk_configurator #1019: [Bug] Should keyboard profile change result in filled profile?

Invertisment, 2021-11-19:
### Describe the Bug

I'm still trying out the web UI and I think there is a bug.
When I choose a keyboard (let's say k320 or eagle_viper) it may have multiple layouts.
And initially it loads fine. But if I change the layout in the dropdown menu then all keys disappear. Is this supposed to happen?

It happens for two keyboards so I think it should happen to all of them.

Reproduction:
1. Open https://config.qmk.fm/#/duck/eagle_viper/v2/LAYOUT_60_ansi or https://config.qmk.fm/#/durgod/k3x0/k320/base/LAYOUT_all
2. Choose a different layout
3. Keymap values disappear

noroadsleft, 2021-11-19:
>  And initially it loads fine (but if I'll use k320tkl then it's a wrong layout). And if I change the layout in the dropdown menu then all keys disappear. Is this supposed to happen?

This is the expected behaviour. QMK Configurator only supports one default keymap per keyboard at this time. Changing the layout doesn't prompt Configurator to look for an appropriate default keymap.

> Also a bad layout is applied when navigating to a page where the keyboard layout is already picked.
> Try opening this in a new tab: [config.qmk.fm/#/durgod/k3x0/k320/base/LAYOUT_tkl_ansi](https://config.qmk.fm/#/durgod/k3x0/k320/base/LAYOUT_tkl_ansi)

This is a long-standing regression of sorts. QMK Configurator used to stay on the layout specified in the URL if you clicked a link that specified both a keyboard and a layout (without searching for a default keymap), but now it loads the default keymap all the time. I don't recall if this is by design, but it's been this way for a few years.

Invertisment, 2021-11-19:
I'm not sure how it's useful. Instead I expected a pre-filled tkl keymap. :/