qmk/qmk_configurator #1025: [Bug] Failed to generate xbows knight (plus) firmware

gang-zhao, 2021-12-04:
### Describe the Bug

Select COMPILE on [xbows knight](https://config.qmk.fm/#/xbows/knight/LAYOUT) or [xbows knight plus](https://config.qmk.fm/#/xbows/knight_plus/LAYOUT) can't generate firmware and the log reported check file size error.
[](url)

### Additional Context?

# KNIGHT
[image]

KNIGHT, A 86 Keys Ergonomic Mechanical Keyboard with RGB backlight.

If you spend your work days in front of a computer, you have likely felt some discomfort after a long day of typing. While spending less time on our computers would be the ideal remedy for this problem, that is not a realistic solution for most people. X-Bows was designed to limit the stress on your hands and wrists while also providing a productive and stylish typing experience.

X-Bows was designed with three criteria in mind:
* Enhancing the comfort and ergonomics of the keyboard
* Limiting the learning curve for our new layout
* Creating an attractive design

* Keyboard Maintainer: X-BOWS
* Hardware Supported: X-BOWS KNIGHT
* Hardware Availability: [X-BOWS](https://x-bows.com/)

Make example for this keyboard (after setting up your build environment):

    make xbows/knight:default
    make xbows/knight:via

Flashing example for this keyboard:

    make xbows/knight:default:flash
    make xbows/knight:via:flash

See the [build environment setup](https://docs.qmk.fm/#/getting_started_build_tools) and the [make instructions](https://docs.qmk.fm/#/getting_started_make_guide) for more information. Brand new to QMK? Start with our [Complete Newbs Guide](https://docs.qmk.fm/#/newbs).

Loaded 2 layers and 172 keycodes. Defined 0 Any key keycodes

Loaded 2 layers and 172 keycodes. Defined 0 Any key keycodes
* Sending xbows/knight:layout_mine with LAYOUT
* Received job_id: e365eb7a-1e37-411b-871a-4a9643e9e74d
* Running . . .
* Finished:
Ψ Compiling keymap with make -s --jobs=1 -r -R -f build_keyboard.mk KEYBOARD=xbows/knight KEYMAP=layout_mine KEYBOARD_FILESAFE=xbows_knight TARGET=xbows_knight_layout_mine KEYBOARD_OUTPUT=.build/obj_xbows_knight KEYMAP_OUTPUT=.build/obj_xbows_knight_layout_mine MAIN_KEYMAP_PATH_1=.build/obj_xbows_knight_layout_mine MAIN_KEYMAP_PATH_2=.build/obj_xbows_knight_layout_mine MAIN_KEYMAP_PATH_3=.build/obj_xbows_knight_layout_mine MAIN_KEYMAP_PATH_4=.build/obj_xbows_knight_layout_mine MAIN_KEYMAP_PATH_5=.build/obj_xbows_knight_layout_mine KEYMAP_C=.build/obj_xbows_knight_layout_mine/src/keymap.c KEYMAP_PATH=.build/obj_xbows_knight_layout_mine/src VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"

avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/xbows/knight/knight.c                                                          
Compiling: .build/obj_xbows_knight_layout_mine/src/keymap.c                                         
Compiling: quantum/quantum.c                                                                        
Compiling: quantum/send_string.c                                                                    
Compiling: quantum/bitwise.c                                                                        
Compiling: quantum/led.c                                                                            
Compiling: quantum/action.c                                                                         
Compiling: quantum/action_layer.c                                                                   
Compiling: quantum/action_macro.c                                                                   
Compiling: quantum/action_tapping.c                                                                 
Compiling: quantum/action_util.c                                                                    
Compiling: quantum/eeconfig.c                                                                       
Compiling: quantum/keyboard.c                                                                       
Compiling: quantum/keymap_common.c                                                                  
Compiling: quantum/keycode_config.c                                                                 
Compiling: quantum/sync_timer.c                                                                     
Compiling: quantum/logging/debug.c                                                                  
Compiling: quantum/logging/sendchar.c                                                               
Compiling: quantum/bootmagic/bootmagic_lite.c                                                       
Compiling: quantum/bootmagic/magic.c                                                                
Compiling: quantum/matrix_common.c                                                                  
[... 26 lines cut ...]
Compiling: tmk_core/protocol/lufa/lufa.c                                                            
Compiling: tmk_core/protocol/usb_descriptor.c                                                       
Compiling: lib/lufa/LUFA/Drivers/USB/Class/Common/HIDParser.c                                       
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Device_AVR8.c                                        
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/EndpointStream_AVR8.c                                
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Endpoint_AVR8.c                                      
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Host_AVR8.c                                          
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/PipeStream_AVR8.c                                    
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Pipe_AVR8.c                                          
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.c                                 
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBInterrupt_AVR8.c                                  
Compiling: lib/lufa/LUFA/Drivers/USB/Core/ConfigDescriptors.c                                       
Compiling: lib/lufa/LUFA/Drivers/USB/Core/DeviceStandardReq.c                                       
Compiling: lib/lufa/LUFA/Drivers/USB/Core/Events.c                                                  
Compiling: lib/lufa/LUFA/Drivers/USB/Core/HostStandardReq.c                                         
Compiling: lib/lufa/LUFA/Drivers/USB/Core/USBTask.c                                                 
Compiling: tmk_core/protocol/lufa/usb_util.c                                                        
Linking: .build/xbows_knight_layout_mine.elf                                                        
Creating load file for flashing: .build/xbows_knight_layout_mine.hex                                
Copying xbows_knight_layout_mine.hex to qmk_firmware folder                                         
Checking file size of xbows_knight_layout_mine.hex                                                 
 *  28880/28672 (208 bytes over)
 

make: *** k:487: check-size] Error 1

gang-zhao, 2021-12-04:
CC @XBowsTech

Lastofthefirst, 2021-12-07:
Same issue, is there a workaround? I'm new, and just  want to change one key D:

fauxpark, 2022-12-13:
Seems to be fixed now.

closed 2022-12-13 by fauxpark