qmk/qmk_configurator #1038: Download Firmware option stays gray (unclickable) on BM60poker_hsrgb_rev2

St4yDown, 2022-01-01:
### Describe the Bug

I've been working on the v2 bm60hsrgb_poker. After compiling a layout it will not allow me to download the firmware. The full source code option works fine but I cant get the firmware to download so I can use QMK Toolbox.

noroadsleft, 2022-01-01:
Appears to be a bug within the `qmk_firmware` repository. Let me dig into this.

noroadsleft, 2022-01-01:
Fixed by https://github.com/qmk/qmk_firmware/pull/15697.

If you have your QMK Configurator keymap.json, open it in a plain text editor (something Notepad, TextEdit, gEdit, or the like), find the line that says `"layout": "LAYOUT",` and change it to `  "layout": "LAYOUT_60_ansi",`, save the file, and import it back into Configurator, and it will compile successfully.

[image]

closed 2022-01-01 by noroadsleft

St4yDown, 2022-01-01:
> Fixed by qmk/qmk_firmware#15697

Sweet! Thank You James!

St4yDown, 2022-01-01:
James,

Thank you so much! Ive been fighting with KPrepublic over these new bm60
v2s for a couple weeks now and Youre the first one to actually help enough
to fix it... I really appreciate this.

Thanks again!
David