qmk/qmk_configurator #1042: Can I use ANY to map such keys combination?

ZaxonXP, 2022-01-05:
### Information

- [X] None of the other issue templates apply

### Description

I was reading the documentation about QMK Configurator, but I still cannot figure out how to program following sequence:

I need to press and hold the * from the keypad, then press 1 from the keypad, then release the *. Can this be done with ANY command?

Kind regards,
Piotr

noroadsleft, 2022-01-05:
This would be a macro command. You can't do this solely in QMK Configurator. What's the use case?

ZaxonXP, 2022-01-05:
I have Keyboard Mouse switch (KM switch) attached to the keyboard and mouse with 4 PC connected. In order to switch between PC's I need to hold asterisk on the keypad and press 1/2/3/4 to select respective PC. However I do not know how to program such macro.

noroadsleft, 2022-01-06:
Okay, you can compile this in QMK Configurator, but it can't be set up fully in Configurator.

## 1. Set up the keymap for your keyboard

Do this as you normally would, then instead of hitting `Compile`, download/export the `keymap.json`.

## 2. Add the macro functionality

When you download the JSON and open it in a text editor, you'll find something like this:

```json
{
  "version": 1,
  "notes": "",
  "documentation": "\"This file is a QMK Configurator export. You can import this at <https://config.qmk.fm>. It can also be used directly with QMK's source code.\n\nTo setup your QMK environment check out the tutorial: <https://docs.qmk.fm/#/newbs>\n\nYou can convert this file to a keymap.c using this command: `qmk json2c {keymap}`\n\nYou can compile this keymap using this command: `qmk compile {keymap}`\"\n",
  "keyboard": "handwired/k_numpad17",
  "keymap": "handwired_k_numpad17_layout_numpad_5x4_mine",
  "layout": "LAYOUT_numpad_5x4",
  "layers": [
    [
      "KC_NLCK",
      "KC_PSLS",
      "KC_PAST",
      "KC_PMNS",
      "KC_P7",
      "KC_P8",
      "KC_P9",
      "KC_P4",
      "KC_P5",
      "KC_P6",
      "KC_PPLS",
      "KC_P1",
      "KC_P2",
      "KC_P3",
      "KC_P0",
      "KC_PDOT",
      "KC_PENT"
    ]
  ],
  "author": ""
}
```

Below the line that says `"layout": "LAYOUT_numpad_5x4",`, add the following:

```json
  "macros": [
    [
      {"action": "down", "keycodes": ["PAST"]},
      {"action": "tap", "keycodes": ["P1"]},
      {"action": "up","keycodes": ["PAST"]}
    ],
    [
      {"action": "down", "keycodes": ["PAST"]},
      {"action": "tap", "keycodes": ["P2"]},
      {"action": "up","keycodes": ["PAST"]}
    ],
    [
      {"action": "down", "keycodes": ["PAST"]},
      {"action": "tap", "keycodes": ["P3"]},
      {"action": "up","keycodes": ["PAST"]}
    ],
    [
      {"action": "down", "keycodes": ["PAST"]},
      {"action": "tap", "keycodes": ["P4"]},
      {"action": "up","keycodes": ["PAST"]}
    ]
  ],
```

Then, somewhere in your keymap, assign the following keycodes:

* `MACRO_0`
* `MACRO_1`
* `MACRO_2`
* `MACRO_3`

These keycodes correspond to the macros defined – `MACRO_0` to the first block (`KC_PAST` down, tap `KC_P1`, release `KC_PAST`), `MACRO_1` to the second, and so on.

## 3. Import and compile

Import the `keymap.json` back into QMK Configurator, and compile as normal.

donhilion, 2022-02-19:
Does this work with the online editor here? https://config.qmk.fm/#//

I tried this to get copy and paste keys.
```json
{
  "layers": [
    [
      "MACRO_0",
      "MACRO_1"
    ]
  ],
  "macros": [
    [
      {"action": "down", "keycodes": ["LCTL"]},
      {"action": "tap", "keycodes": ["C"]},
      {"action": "up","keycodes": ["LCTL"]}
    ],
    [
      {"action": "down", "keycodes": ["LCTL"]},
      {"action": "tap", "keycodes": ["V"]},
      {"action": "up","keycodes": ["LCTL"]}
    ]
  ]
}
```

After importing, compiling and flashing the keys don't seem to do anything.

noroadsleft, 2022-02-19:
@donhilion Yours can be done with the native keycodes present on QMK Configurator:

[image]

Drag the `LCTL()` keycode (`Quantum` tab under `Mod key combinations`) onto the desired keys, then drag C and V, respectively, into the inner boxes on the keymap.

donhilion, 2022-02-24:
That works, thanks!
But there remains the question: What is wrong with this configuration or the steps taken?

Esnos33, 2023-03-09:
@noroadsleft Hi, I have problem with my macro. I wanted to have macro to make "->" arrow, but it doesn't work. I modified my json file like this
```
"keymap": "esnos",
  "macros": [
    [
      {"action": "tap", "keycodes": ["KC_MINUS", "KC_GT"]}
    ]
  ],
  "layout": "LAYOUT_split_3x5_2",
  "layers": [
    [
      "KC_ESC",
```
and later in code I have:
```
      "KC_ENT",
      "QK_MACRO_0",
      "KC_LPRN",
      "KC_RPRN",
```
and in qmk configurator uploaded json file looks like this:
https://imgur.com/a/zaABNMJ
but it doesn't work. Also, if I use MACRO_0 instead of QK_MACRO_0, I have this compilation error:
gmake: *** ap_introspection.o] Error 1

After uploading firmware to my keyboard, this button doesn't work, it doesn't send any signal according to KeyboardStateView.

Could you please help me? I can't find anybody on internet who have macro with json file.

noroadsleft, 2023-03-09:
@Esnos33,

For your tap action, try this:

```
       {"action": "tap", "keycodes": ["MINS"]},
       {"action": "tap", "keycodes": ["GT"]}
```

Esnos33, 2023-03-09:
I have something like this:
```
  "macros": [
    [
      {"action": "tap", "keycodes": ["MINS"]},
      {"action": "tap", "keycodes": ["GT"]}
    ]
```
and QK_MACRO_0 in code, but still it doesn't work

Jefferycheng, 2023-03-11:
@Esnos33 

I want to input "=>" through one key, the same idea as you.
I thought this was a bug of qmk configurator.
I directly use the qmk firmware to compile that works!
The following are my macros. 

```
"macros": [
    [
      {
        "action": "tap",
        "keycodes":["EQL"]
      },
      {
        "action": "down",
        "keycodes":["LSFT"]
      },
      {
        "action": "tap",
        "keycodes":["DOT"]
      },{
        "action": "tap",
        "keycodes":["LSFT"]
      }
    ]
  ]
```

The qmk firmware overview is here : https://docs.qmk.fm/#/newbs?id=overview

Esnos33, 2023-03-11:
@Jefferycheng Ok, I will try directly compile firmware and see what happens, thanks for answer.

Esnos33, 2023-03-11:
@Jefferycheng I must admint, whole process is very hard just to have this one arrow, I will use other software to have this arrow

Jefferycheng, 2023-03-13:
@Esnos33 I agree with you the whole process is long and complex.

Even though I have run the process before, I often forget how to do this.

So now, I am recording the whole process in my blog here: https://medium.com/p/9b9bc0bad275

Wish this can help you.

Esnos33, 2023-03-13:
@Jefferycheng It works! Thanks for help, I really appreciate it.

Nairda015, 2023-09-21:
I've conducted some tests on my end. Here are the observations:

- When compiled locally, the macro works as expected.
- When the same file is uploaded to https://config.qmk.fm/, the compilation succeeds, but the keyboard does not recognize the macro.

Sidenote:  
@Jefferycheng, the QMK [documentation](https://docs.qmk.fm/#/newbs_building_firmware) provides a tip on simplifying the process. It suggests assigning a default keyboard and keymaps. This allows you to run the following command: `QMK compile -e BUILD_DIR="./builds"`

Additional Information:  
The entire local build process took me approximately 10 minutes. However, installing QMK from Homebrew took over an hour, which I'm not sure is typical behaviour.

marcoXbresciani, 2024-09-30:
Same issue for me here, right after one year.
I'm using online Configurator and it fits me very well, with (at the moment?) absolutely no need to go with the code.

I'm trying to add 6 macros to write capital accented letters we sometimes use here in Italy: `À` (Alt+0192), `È` (Alt+0200), `É` (Alt+0201), `Ì` (Alt+0204), `Ò` (Alt+0210), `Ù` (Alt+0217).
Same process as above, my macros are:
```
  "macros": [
    [
        { "action": "down", "keycodes": [ "KC_LALT" ] },
        { "action": "tap", "keycodes": [ "KC_P0" ] },
        { "action": "tap", "keycodes": [ "KC_P1" ] },
        { "action": "tap", "keycodes": [ "KC_P9" ] },
        { "action": "tap", "keycodes": [ "KC_P2" ] },
        { "action": "up", "keycodes": [ "KC_LALT" ] }
    ],
    [
        { "action": "down", "keycodes": [ "KC_LALT" ] },
        { "action": "tap", "keycodes": [ "KC_P0" ] },
        { "action": "tap", "keycodes": [ "KC_P2" ] },
        { "action": "tap", "keycodes": [ "KC_P0" ] },
        { "action": "tap", "keycodes": [ "KC_P0" ] },
        { "action": "up", "keycodes": [ "KC_LALT" ] }
    ],
    [
        { "action": "down", "keycodes": [ "KC_LALT" ] },
        { "action": "tap", "keycodes": [ "KC_P0" ] },
        { "action": "tap", "keycodes": [ "KC_P2" ] },
        { "action": "tap", "keycodes": [ "KC_P0" ] },
        { "action": "tap", "keycodes": [ "KC_P1" ] },
        { "action": "up", "keycodes": [ "KC_LALT" ] }
    ],
    [
        { "action": "down", "keycodes": [ "KC_LALT" ] },
        { "action": "tap", "keycodes": [ "KC_P0" ] },
        { "action": "tap", "keycodes": [ "KC_P2" ] },
        { "action": "tap", "keycodes": [ "KC_P0" ] },
        { "action": "tap", "keycodes": [ "KC_P4" ] },
        { "action": "up", "keycodes": [ "KC_LALT" ] }
    ],
    [
        { "action": "down", "keycodes": [ "KC_LALT" ] },
        { "action": "tap", "keycodes": [ "KC_P0" ] },
        { "action": "tap", "keycodes": [ "KC_P2" ] },
        { "action": "tap", "keycodes": [ "KC_P1" ] },
        { "action": "tap", "keycodes": [ "KC_P0" ] },
        { "action": "up", "keycodes": [ "KC_LALT" ] }
    ],
    [
        { "action": "down", "keycodes": [ "KC_LALT" ] },
        { "action": "tap", "keycodes": [ "KC_P0" ] },
        { "action": "tap", "keycodes": [ "KC_P2" ] },
        { "action": "tap", "keycodes": [ "KC_P1" ] },
        { "action": "tap", "keycodes": [ "KC_P7" ] },
        { "action": "up", "keycodes": [ "KC_LALT" ] }
    ]
  ],
```
because I have to use Alt+numpad numbers to write them, with Italian (142) layout on my (Windows) PC since they are not present on the keyboard.
Compiles perfectly, but no key sent.

Hints for me? Shall I move to code? :(

marcoXbresciani, 2024-10-09:
Damn, stupid me!
It's because macros don't use `KC_` in the keycodes. 🙄