qmk/qmk_configurator #1046: [Bug] Keyboard Hillside is not loading its default keymap

mmccoyd, 2022-01-13:
### Describe the Bug

Hillside has a default keymap in hardwired/hillside/keymaps/default/keymap.json, yet it is not loaded when the keyboard is selected nor when the default button is pressed.

It does load successfully if done so with the upload JSON button.

### Additional Context?

Hillside was merged into QMK with https://github.com/qmk/qmk_firmware/pull/15399#ref-commit-ede43cd on January 8.

yanfali, 2022-01-13:
Doesn't work that way I'm afraid. Keymaps aren't loaded from the repo for technical reasons. @noroadsleft

mmccoyd, 2022-01-13:
Ok. I did not realize. The keyboard and its layout got from qmk to the configurator. 

How to I trigger the layout to get there as well?

yanfali, 2022-01-13:
Someone needs to add a default keymap for configurator. I'm sure noroads has it in hand.

noroadsleft, 2022-05-27:
Solved by #1074.

closed 2022-05-27 by noroadsleft