qmk/qmk_configurator #1081: Chatter detection message question

hudson4351, 2022-02-16:
### Information

- [X] None of the other issue templates apply

### Description

I suspect the '6' key on one of my keyboards is chattering. I used the test page here:

https://config.qmk.fm/#/test

And obtained the following output when pressing the '6' key once followed by the '1' key once:

[image]

Is the above output indicative of chattering? Is there supposed to be something on the page that alerts me when chattering occurs other than manual review of the console?

When I look at this post, I see the text "CHATTER HAS BEEN DETECTED!" displayed next to the chatter threshold window:

https://github.com/qmk/qmk_configurator/pull/466#issue-480032675

Is that text always supposed to be displayed whenever chattering occurs?

yanfali, 2022-02-17:
Hi, I don't see any evidence of chatter. The UI defines chatter as a repeat key event within 8ms. I'm not seeing that one your trace.

hudson4351, 2022-02-17:
The UI doesn't show the amount of time that has elapsed from each KEY-UP event to the next KEY-DOWN event, but that is the number that is being compared against the chatter threshold, correct? The UI only shows the amount of time that has elapsed from each KEY-DOWN event to the next KEY-UP event.

Where does the default value of "8 ms" come from? I definitely only pressed the '6' key once but it looks like two presses were recorded. I now see why that didn't display the "CHATTER HAS BEEN DETECTED!" alert, but another thing I noticed is that if I run the same test using Elite Keyboard's Switch Hitter tool with the same chatter threshold of 8 ms, I am able to fairly consistently trigger its "CHATTER DETECTED" alert:

[image]

yet I am not able to trigger the "CHATTER HAS BEEN DETECTED!" alert on the QMK webpage no matter how many times I press the same '6' key. Could there be a latency issue with trying to measure keyboard switch delays through a website instead of a local app?

hudson4351, 2022-02-18:
Is there any particular reason "8 ms" is the default value?

> this is definitely a best effort kind of tool and not meant to be as definitive as switch hitter which is a native program. It could be swapping out to disk or your machine might be slow, or you could be bit coin mining. All these things will affect latency on a windows PC.

Does my latency to the website affect the delay measurements even if no plugins are running?

yanfali, 2022-02-18:
You would have to ask @zekth who wrote the feature. I think it's just a
reasonable value chosen to reflect a threshold where it is unreasonable to
expect a 2nd keypress. Keyboards usually denounce each press, so getting
one this quickly is often symptomatic of a hardware issue. We are making a
best effort in software to diagnose hardware. If you notice, it's
configurable, so choose whatever value you want. This is just work works
for most people.

Performance in general is hard to quantify. What's fast for you may be slow
for others. Just be aware that your browser is running on top of many
layers of software and each one is adding some measurable amount of delay.
It's just hard to quantify on a browser since it's at the tail end.

Hope this helps. Please feel free to use whatever tool gets you to a
reasonable result. We make no guarantees about the fitness of purpose for
chatter detection. We just provided it because it seemed useful and switch
hitter is no longer maintained.

zekth, 2022-02-18:
> Is there any particular reason "8 ms" is the default value?
>
>
> Does my latency to the website affect the delay measurements even if no plugins are running?

It does not, the code is executed in the page. This feature has been tested with a chattery keyboard but yeah it cannot be as reliable as switch hitter because of the toolchain involved.

Regarding the 8ms it's the default of switch hitter.