qmk/qmk_configurator #1083: [Feature Request] Generate a cheatsheet image

octref, 2022-02-18:
### Description

Hi there - love this project & thanks for your hard work!

I'd like to get a cheatsheet of my layout each time I update it. Found this on reddit:

[image]
 
Ideally, from a keymap.json, I get an image showing what's on layer 1/3/4/6.

One way of doing this would be rendering the layout to DOM, and use something like https://github.com/tsayen/dom-to-image to capture it.
I might be able to send a PR, if the maintainers see this as a worthwhile addition.

yanfali, 2022-02-18:
Have you tried printing? There's a print button

octref, 2022-02-18:
Just saw that, while it's a nice feature, I'd still love to have all info in one image...

yanfali, 2022-02-18:
It's pretty difficult to fit all that in one keyboard. What if you had 10
layers?

octref, 2022-02-18:
Yeah I don't think it'd apply to all use-cases, that's why I'm hesitant if this is a good addition to the configurator. I do guess 95% of the people use the 3 to 4 layers that's factory default, though. To those people this could be super useful.

octref, 2022-02-18:
Seems some people already gave it a shot:
- https://github.com/brianoverby/OrthoKeymapSVG
- https://www.reddit.com/r/olkb/comments/oll0oi/create_you_own_planckpreonic_keymap/

octref, 2022-02-18:
ok thanks!

closed 2022-02-18 by octref