qmk/qmk_configurator #1084: [Feature Request] Option to change Tapping Term default in Configurator

writezhe, 2022-02-24:
### Description

Hi, I'm new the QMK software and vocabulary, so apologies for any mistaken terms. I've looked through the prior requests and didn't see this.

I think the use of MT and LT tap-hold keys are great. However, I notice some timing issues using it (which can be personal, as noted in the [Tap-Hold docs page](https://docs.qmk.fm/#/tap_hold?id=tapping-term)). 

I'm not comfortable yet writing my own configuration file (I will try to learn over the next few weeks to try to replicate my QMK Configurator version), but if feasible, I'd like to be able to adjust the TAPPING_TERM option (or other relevant tap-hold options) via the Configurator.

Thanks!

yanfali, 2022-02-24:
@skullydazed

jorgen99, 2022-04-07:
+1
I added "Any" keys with "DT_PRNT", "DT_UP" and "DT_DOWN" but I can't get it to work. So, a setting for this would be great.