qmk/qmk_configurator #1085: Montex Keyboard Cannot Load Layout

thommattson, 2022-02-26:
### Describe the Bug

This bug applies to the Idobao Montex -- keyboard "Montex" in QMK Configurator.

Two weeks ago (Feb 11) I was able to load the default layout "numpad_6x5" in QMKC, compile the code, download the JSON, and write it to firmware. Supper easy; no issues. As of today (Feb 25) the Montex appears not to be a supported keyboard. See screenshot. All other keyboard make/models load without issue. 

Please help! Thanks.

[image]

yanfali, 2022-02-26:
Looks like someone moved it, so your URL is wrong
https://config.qmk.fm/#/idobao/montex/v1/LAYOUT_numpad_6x5

thommattson, 2022-02-26:
> Looks like someone moved it, so your URL is wrong https://config.qmk.fm/#/idobao/montex/v1/LAYOUT_numpad_6x5

Thanks for the direct link! Saw it was moved but couldn't pinpoint where. Not populating the front-end drop down.

Appreciate it!

closed 2022-02-26 by yanfali