qmk/qmk_configurator #1092: [Bug] Drop Alt - QMK configurator generated .bin is bricking the keyboard?

skulltraill, 2022-03-01:
### Describe the Bug

After creating a config for drop alt on QMK configurator, compiling, and downloading the `.bin`, and flashing it using QMK toolbox, the keyboard dies, no RGB, no keypresses, it's basically bricked, until I reflash with an older or the default `.bin` which restores it.

This lead me to believe that the `.bin`s from older QMK configurator compilations are working fine, but not the current one on config.qmk.fm

I know the QMK configurator prints out:
```@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
This MCU support package has a lack of support from the upstream provider (Massdrop).
There are currently questions about valid licensing, and at this stage it's likely
their boards and supporting code will be removed from QMK in the near future. Please
contact Massdrop for support, and encourage them to align their future board design
choices to gain proper license compatibility with QMK.
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@```

So this is likely not your fault, nor am I expecting a fix, just wanted to bring this to your attention.

Thank you!

### Additional Context?

- Keyboard:  Drop Alt
- Operating system: Windows 10

yanfali, 2022-03-01:
Any insight into this @zvecr ?

zvecr, 2022-03-02:
Going to tackle this one over at https://github.com/qmk/qmk_firmware/issues/16489 as we dont need 2 tickets for the same thing.

closed 2022-03-02 by zvecr