qmk/qmk_configurator #1099: [Bug] Error compiling keymap for kinesis/kint41

schellj, 2022-03-09:
### Describe the Bug

Attempting to compile a keymap for the kinesis/kint41 board results in the error below and a failed build. As of this writing, this works fine using qmk/qmk_firmware:master directly.

```
# kinesis_kint41 keyboard firmware

Please see https://github.com/kinx-project/kint for details.

Loaded 1 layers and 86 keycodes. Defined 0 Any key keycodes
* Sending kinesis/kint41:basic-customizations with LAYOUT
* Received job_id: 97cf94fa-b0da-4090-b720-85f031718955
* Running . . . . . . . . .
* Finished:
Ψ Compiling keymap with make -s --jobs=1 -r -R -f builddefs/build_keyboard.mk KEYBOARD=kinesis/kint41 KEYMAP=basic-customizations KEYBOARD_FILESAFE=kinesis_kint41 TARGET=kinesis_kint41_basic-customizations KEYBOARD_OUTPUT=.build/obj_kinesis_kint41 KEYMAP_OUTPUT=.build/obj_kinesis_kint41_basic-customizations MAIN_KEYMAP_PATH_1=.build/obj_kinesis_kint41_basic-customizations MAIN_KEYMAP_PATH_2=.build/obj_kinesis_kint41_basic-customizations MAIN_KEYMAP_PATH_3=.build/obj_kinesis_kint41_basic-customizations MAIN_KEYMAP_PATH_4=.build/obj_kinesis_kint41_basic-customizations MAIN_KEYMAP_PATH_5=.build/obj_kinesis_kint41_basic-customizations KEYMAP_C=.build/obj_kinesis_kint41_basic-customizations/src/keymap.c KEYMAP_PATH=.build/obj_kinesis_kint41_basic-customizations/src VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"

arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) bedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/kinesis/kinesis.c                                                             In file included from ./lib/chibios/os/common/ports/ARMCMx/chcore.h:70:0,
                 from ./lib/chibios/os/rt/include/ch.h:112,
                 from ./lib/chibios/os/hal/osal/rt-nil/osal.h:32,
                 from ./lib/chibios/os/hal/include/hal.h:28,
                 from platforms/chibios/platform_deps.h:18,
                 from quantum/quantum.h:18,
                 from keyboards/kinesis/kint41/kint41.h:19,
                 from keyboards/kinesis/kinesis.h:14,
                 from keyboards/kinesis/kinesis.c:1:
./lib/chibios-contrib/os/common/startup/ARMCMx/devices/MIMXRT1062/cmparams.h:70:24: fatal error: MIMXRT1062.h: No such file or directory
 #include "MIMXRT1062.h"
                        ^
compilation terminated.
 
 | 
 | 
 | 
make: *** izations/keyboards/kinesis/kinesis.o] Error 1
* Sending kinesis/kint41:basic-customizations with LAYOUT
* Received job_id: 236c546f-a5c7-4109-9ca6-8f309e0240a3
* Running . . . . . . . . . . .
* Finished:
Ψ Compiling keymap with make -s --jobs=1 -r -R -f builddefs/build_keyboard.mk KEYBOARD=kinesis/kint41 KEYMAP=basic-customizations KEYBOARD_FILESAFE=kinesis_kint41 TARGET=kinesis_kint41_basic-customizations KEYBOARD_OUTPUT=.build/obj_kinesis_kint41 KEYMAP_OUTPUT=.build/obj_kinesis_kint41_basic-customizations MAIN_KEYMAP_PATH_1=.build/obj_kinesis_kint41_basic-customizations MAIN_KEYMAP_PATH_2=.build/obj_kinesis_kint41_basic-customizations MAIN_KEYMAP_PATH_3=.build/obj_kinesis_kint41_basic-customizations MAIN_KEYMAP_PATH_4=.build/obj_kinesis_kint41_basic-customizations MAIN_KEYMAP_PATH_5=.build/obj_kinesis_kint41_basic-customizations KEYMAP_C=.build/obj_kinesis_kint41_basic-customizations/src/keymap.c KEYMAP_PATH=.build/obj_kinesis_kint41_basic-customizations/src VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"

arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) bedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/kinesis/kinesis.c                                                             In file included from ./lib/chibios/os/common/ports/ARMCMx/chcore.h:70:0,
                 from ./lib/chibios/os/rt/include/ch.h:112,
                 from ./lib/chibios/os/hal/osal/rt-nil/osal.h:32,
                 from ./lib/chibios/os/hal/include/hal.h:28,
                 from platforms/chibios/platform_deps.h:18,
                 from quantum/quantum.h:18,
                 from keyboards/kinesis/kint41/kint41.h:19,
                 from keyboards/kinesis/kinesis.h:14,
                 from keyboards/kinesis/kinesis.c:1:
./lib/chibios-contrib/os/common/startup/ARMCMx/devices/MIMXRT1062/cmparams.h:70:24: fatal error: MIMXRT1062.h: No such file or directory
 #include "MIMXRT1062.h"
                        ^
compilation terminated.
 
 | 
 | 
 | 
make: *** izations/keyboards/kinesis/kinesis.o] Error 1
```

fauxpark, 2022-03-09:
Duplicate of https://github.com/qmk/qmk_firmware/issues/16440

closed 2022-03-09 by fauxpark

schellj, 2022-03-09:
Sorry for the duplicate issue; I'm not very familiar with this project and didn't realize the proper place for this issue to live would be in qmk/qmk_firmware, so I didn't look there first.