qmk/qmk_configurator #1110: How to add a new keyboard item in Dropdown. Thank you very much

SEAN-JOU, 2022-04-14:
### Description

[image]

### What do i prepare something to add it? Thank you very much

noroadsleft, 2022-04-14:
QMK Configurator only supports keyboards with support merged in to [qmk/qmk_firmware's `master` branch](https://github.com/qmk/qmk_firmware/tree/master/).

Once you've done that, the instructions for adding a default keymap to QMK Configurator are here:

* [Adding Default Keymaps - QMK Firmware Documentation](https://docs.qmk.fm/#/configurator_default_keymaps)

SEAN-JOU, 2022-04-18:
I am a keyboard maker.I want to add QMK firmware and  join the project for merging master.

nooges, 2022-04-18:
Please see https://docs.qmk.fm/#/contributing

closed 2022-06-29 by nooges