qmk/qmk_configurator #1124: [Bug] GMMK Pro keymaps missing

lukini, 2022-05-29:
### Describe the Bug

If you select either GMMK Pro keyboard from the dropdown, there is no default keymap.  I believe this is because the existing maps (such as public/keymaps/g/gmmk_pro_ansi_default.json) are referencing the old path when the new ones have /rev1/ added to them.

egreen0, 2022-05-31:
I am running into the same issue.

noroadsleft, 2022-05-31:
Apologies for this. 

I haven't completed the branch that relocates the keymaps for the keyboards that have moved with this most recent `develop` branch merge. I should have it completed within a day or two.

I'm running behind because I was unwell all of last week.

noroadsleft, 2022-06-01:
Confirming this issue is now fixed.

PR #1128 (2022-06-01): Update Default Keymaps and Redirect Table: QMK 0.17.0
Updates the keymaps for keyboards that have moved in QMK with the release of 0.17.0.

Fixes #1124.
closed 2022-06-01 by noroadsleft