qmk/qmk_configurator #1136: [Bug] redox_w's firmware not working whan reconnect the receiver

mffseal, 2022-06-03:
### Describe the Bug

Precompiled receiver firmware warks fine, but the firmware downloaded form qmk_configurator does not work whan reconnect the receiver.

I must burn the precompiled firmware first and and then the downloaded firmware to work, but once the receiver is restarted(power off or disconnect from my pc), keyboard not responding.

yanfali, 2022-06-03:
Qmk doesn't support wireless boards except bluetooth at this time. So not sure how this could work.

https://github.com/mattdibi/redox-keyboard#wireless-implementations

Use zmk or out of tree firmware. Mainline has no support at this time.

sigprof, 2022-06-03:
Actually the underlying bug was reported at qmk/qmk_firmware#16553, and there is even a suggested fix at qmk/qmk_firmware#17203 — once the fix is merged into the master branch, it would be available in the QMK Configurator too. Please watch the status of that bug instead.

closed 2022-06-03 by sigprof