qmk/qmk_configurator #1143: [Bug] "Menu" key should probably use KC_APP instead of KC_MENU (or, have both as options)

mgsloan, 2022-06-20:
### Describe the Bug

When I use the "Menu" button from the media keys section, it does not work as expected.  Instead `xev` shows it is keycode 138 "SunProps" instead of keycode 135 "Menu".  Manually modifing the json file to use `KC_MENU` fixes the issue.

From https://www.reddit.com/r/olkb/comments/5uzhiq/whats_the_difference_between_kc_app_and_kc_menu/

> All the search results say it does unexpected stuff and that kc_app is the code for the menu key.
>
> I am curious as well though because I haven't seen a good explanation of what KC_MENU should be for.

See also [KC_MENU produces Sun Props key · Issue #306 · qmk/qmk_firmware](https://github.com/qmk/qmk_firmware/issues/306)

I propose that QMK configurator instead associate `KC_APP` with the media key called "Menu"

### Additional Context?

I like to have a key on my keyboard to open right-click context menus.  I discovered this was possible while reading the documentation for the text format used for configuring kinesis advantage 2 keyboard vanilla firmware.  One of my kinesii controllers broke, so I replaced it with [KinT](https://github.com/kinx-project/kint) + QMK.

In the Kinesis text configuration language the key I want is called `[menu]` (also yielded by `KC_APP`) and `xev` shows it to be

```
KeyPress event, serial 33, synthetic NO, window 0x3e00001,
    root 0x7a4, subw 0x0, time 39243766, (367,300), root:(1553,300),
    state 0x10, keycode 135 (keysym 0xff67, Menu), same_screen YES,
```

Whereas the menu media key in the configurator / `KC_MENU` in configurator json results in

```
KeyPress event, serial 33, synthetic NO, window 0x3e00001,
    root 0x7a4, subw 0x0, time 39863037, (367,200), root:(1553,600),
    state 0x10, keycode 138 (keysym 0x1005ff70, SunProps), same_screen YES,
```

mgsloan, 2022-06-20:
Ahah!  https://github.com/qmk/qmk_configurator/pull/770

> KC_APP now uses a Hamburger Menu icon to differentiate it visually from KC_MENU. The latter is now marked as "(Legacy)" in its help text.

Uhhh that seems like a strange way to handle this problem.  I would never guess that menu == hamburger.  The hamburger icon also does not appear in the configurator.

I tried to fix this, but I'm not sure how to move forward nicely with this hamburger convention, so it seems there are decisions to be made.  Personally I think the following is good:

* `KC_MENU` as display for the legacy binding.
* `Menu` as display for `KC_APP`

yanfali, 2022-06-21:
As a work around try ANY(KC_APP)

mgsloan, 2022-06-26:
It turns out it is possible in the configurator UI.  For key selection there are two keys with the text "Menu", one on the ansi selector pane (which uses `KC_APP`, which I think is correct), but it renders differently when deployed - instead using the hamburger icon.  On the media key selection pane there is also a key with the text "Menu", but it does not work as expected.

noroadsleft, 2022-06-27:
@mgsloan How do you feel about the following change?

`KC_APP` (unchanged):
[image]

`KC_MENU` (marked as `(Legacy)`):
[image]

noroadsleft, 2022-06-28:
I'd actually be more likely to revert the hamburger icon. I added that originally because users weren't able at the time to differentiate between the modern Menu key (`KC_APP`) and the legacy `KC_MENU` in the keymap display. I'm hesitant to use icons in the keycode select area because for the most part, the icons aren't standardized, and the ones that are are not widely used/known.

closed 2022-06-28 by noroadsleft

mgsloan, 2022-06-28:
> completed via #1145 <https://github.com/qmk/qmk_configurator/pull/1145>.

Yes, exactly what I was thinking too!

Thanks so much for addressing this, I hope it'll reduce confusion for
others in the future (my keyboard lacked a functioning menu button for
quite a while, which I'm accustomed to having on my unmodified Kinesis)

I highly recommend trying out including this key in layouts, btw, it is
quite useful.