qmk/qmk_configurator #1172: Old JSON is not read

nysvensson, 2022-08-26:
### Describe the Bug

When I upload a JSON created in 2020 with QMK, nothing renders

{"version":1,"notes":"","documentation":"\"This file is a QMK Configurator export. You can import this at <https://config.qmk.fm>. It can also be used directly with QMK's source code.\n\nTo setup your QMK environment check out the tutorial: <https://docs.qmk.fm/#/newbs>\n\nYou can convert this file to a keymap.c using this command: `qmk json2c {keymap}`\n\nYou can compile this keymap using this command: `qmk compile {keymap}`\"\n","keyboard":"ergodox_ez","keymap":"ergodox_ez_3","layout":"LAYOUT_ergodox_pretty","layers":[["KC_ESC","KC_1","KC_2","KC_3","KC_4","KC_5","KC_PSLS","KC_QUOT","KC_6","KC_7","KC_8","KC_9","KC_0","KC_SLSH","KC_TAB","KC_Q","KC_W","KC_E","KC_R","KC_T","KC_LBRC","KC_RBRC","KC_Y","KC_U","KC_I","KC_O","KC_P","KC_EQL","LCTL(KC_V)","KC_A","KC_S","KC_D","KC_F","KC_G","KC_H","KC_J","KC_K","KC_L","KC_SCLN","KC_BSLS","LCTL(KC_C)","LCTL_T(KC_Z)","KC_X","KC_C","KC_V","KC_B","KC_ASTR","KC_LPRN","KC_N","KC_M","KC_COMM","KC_DOT","KC_NUBS","LCTL(KC_F)","LCTL(KC_X)","KC_LGUI","KC_LCTL","KC_LEFT","KC_RGHT","KC_UP","KC_DOWN","KC_MINS","KC_GRV","MO(1)","KC_DEL","KC_CAPS","KC_RALT","KC_RSFT","KC_HOME","KC_PGUP","KC_BSPC","KC_LSFT","KC_END","KC_PGDN","KC_ENT","KC_SPC"],["KC_NO","KC_F1","KC_F2","KC_F3","KC_F4","KC_F5","KC_TRNS","KC_TRNS","KC_F6","KC_F7","KC_F8","KC_F9","KC_F10","KC_F11","KC_TRNS","KC_EXLM","KC_AT","KC_LCBR","KC_RCBR","KC_PIPE","KC_TRNS","KC_TRNS","KC_UP","KC_7","KC_8","KC_9","KC_ASTR","KC_F12","KC_TRNS","KC_HASH","KC_DLR","KC_LPRN","KC_RPRN","KC_GRV","KC_DOWN","KC_4","KC_5","KC_6","KC_PLUS","KC_TRNS","KC_TRNS","KC_PERC","KC_CIRC","KC_LBRC","KC_RBRC","KC_TILD","KC_TRNS","KC_TRNS","KC_AMPR","KC_1","KC_2","KC_3","KC_BSLS","KC_TRNS","EEP_RST","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_DOT","KC_0","KC_EQL","OSL(0)","RGB_MOD","KC_TRNS","RGB_TOG","KC_NO","KC_TRNS","KC_TRNS","RGB_VAD","RGB_VAI","KC_TRNS","KC_TRNS","RGB_HUD","RGB_HUI"],["KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_MS_U","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_MS_L","KC_MS_D","KC_MS_R","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_MPLY","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_MPRV","KC_MNXT","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_BTN1","KC_BTN2","KC_VOLU","KC_VOLD","KC_MUTE","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_WBAK"]],"author":""}

zvecr, 2022-08-26:
Fixed, now imports correctly.

[image]

closed 2022-08-26 by zvecr

nysvensson, 2022-08-27:
Amazing, thank you!