qmk/qmk_configurator #1179: [Bug] Microdox keyboards not laid out correctly

timtraversy, 2022-08-31:
### Describe the Bug

The boardsource/microdox/v1 and v2 keyboards are displayed in one long row, instead of their proper 3x5 layout. I can't see where in the source code the geometry is actually defined.

[image]

yanfali, 2022-08-31:
Qmk_firmware keyboard info.json

timtraversy, 2022-08-31:
Thank you - I would expect the bug is originating from there, but the microdox file hasn't been updated since July and this is a new issue.

timtraversy, 2022-08-31:
Ok, so the v2 microdox defines its own info.json, but the Configurator is fetching the one from the root of the microdox directory. Is it not supported for vX versions to override the geometry?

noroadsleft, 2022-08-31:
This is actually an new bug in our infrastructure with how QMK Firmware merges multiple `info.json` files together. All the requisite data is present already. I flagged this internally a couple of days ago, but I'm not sure where we stand on this.

I could patch it, but I'd rather have a proper fix done.

keyboard-magpie, 2022-08-31:
It's quite interesting, when you preview the info.json in configurator with ctrl-shift-i, you briefly see the correct layout for a second before it transforms to the long line.

timtraversy, 2022-08-31:
I mistakenly though the v2 board was overwriting the layout, but it just defines its own RGB matrix. I'm pretty sure this is just because key positions were never added to the microdox `info.json`, so I'll close this

closed 2022-08-31 by timtraversy