qmk/qmk_configurator #1227: [Bug] planck/rev6 compiling

Carlos-918, 2022-12-14:
### Describe the Bug

When ever trying to compile in the online configurator a planck/rev6 or planck/rev6_drop, an error(Firmware size check does not yet support STM32F303; skipping) appears. Even in the qmk default layout. No sure what is happening, maybe its me, but is kind of anoying that all the others work fine >:(  Just some days ago I used it and was perfectly fine. I don't know what is happening now. Please help :(

tzarc, 2022-12-14:
That's not an error -- it'd work under normal circumstances.
There's a bug for all ARM builds -- with a fix incoming -- just need to sit tight while we sort it out.

tzarc, 2022-12-14:
Should be fine, now. Can you please confirm?

Carlos-918, 2022-12-14:
> Should be fine, now. Can you please confirm?

Yes, it works fine again. Thank you so much for the quick response.

closed 2022-12-14 by Carlos-918