qmk/qmk_configurator #1242: [Feature Request] i would like to be able to run this thing offline

colons, 2023-02-21:
### Description

it is very strange to me that this tool for creating keyboard firmware has network dependencies that seem very hard to remove. i spent a few hours trying to get [the whole stack](https://docs.qmk.fm/#/configurator_architecture) running locally, but had very little success

i have some motivations for wanting to do this:

- i want to be able to edit my keyboard firmware when i don't have internet access
- i do not trust any of the services in that architecture diagram to be around forever
- i do not want to have my keyboard layouts stored on other people's computers

i feel like these are all reasonable desires, and kinda fundamental to tinkery open source stuff in general

my ideal version of this would be an entirely self-contained desktop app with a web view that exposes the existing configuration gui without hitting the network at all. after all, if being able to run this thing offline is a desirable thing, then it should be as accessible as possible

if it is already possible to run the stack locally, i would really like there to be documentation that covers how. i would be willing to write it myself, given guidance to understand what's necessary

fauxpark, 2023-02-21:
https://github.com/qmk/qmk_web_stack is probably what you're looking for.

marschr, 2023-03-13:
I just tried running the qmk_web_stack repo and it fails because it seems to have dependencies with a AWS server... `urllib3.exceptions.NewConnectionError: <botocore.awsrequest.AWSHTTPConnection object at 0x7f21e505e760>: Failed to establish a new connection: [Errno 111] Connection refused`
The configurator architecture page helps a bit to understand how everything is tied together but may I ask if there's any low hanging fruit advice on where to decouple these containers in order to have a locally ran configurator, what is actually needed and for what, so I could load and customize my own JSON keymap files using the WebUI? Also, shouldn't the AWS dependencies be moved to a setting in an env file? Or am I missing something?

tzarc, 2023-03-13:
The recommendation is to use the `docker-compose.yml` to start the services -- the storage of files uses MinIO, which is an S3-compatible service. I'd imagine you haven't started it?

I've run through the instructions as per the `qmk_web_stack` readme, and the services come up fine.

I realise you may wish to bundle things up into your own local desktop app, but realistically, that's the premise of having QMK installed locally and compiling with the CLI.

marschr, 2023-03-14:
I managed to run it. Somehow lots of request from all the docker-compose services starting at the same time was timing out some of the docker images retrieval. It was probably my cable modem connection wasn't able to handle all the concurrent connections, building all the `qmk_web_stack` compose services one by one helped.

Now for the real question, on the line 408 in the file `web.py` from the `web_api` repo ( https://github.com/qmk/qmk_api/blob/b2fe28f4cc26e3b23d0ba053e13cc1ada9439df1/web.py#L408 ) there's a redirect to the `keyboards.qmk.fm` API , is that by design? Since my objective is to tune my custom firmware locally I can't access my local .json file without changing those remote API calls, which wouldn't be an issue for a hacky one time solution on my end. Thanks!

Nazosan, 2025-08-04:
I'm afraid that I don't.  I merely know *of* the software, not how to use it to do this.

That's why this is a suggestion, not a PR.

fauxpark, 2025-08-04:
> As mentioned above, there are issues like the lack of guarantee the services will always be around

You've misread what happened above. The web stack services don't talk to any Amazon servers at all, it has its own S3 replacement (minio) but the API and libraries used are the same so the error messages will still reference AWS.

> Certainly doing a whole docker just for QMK Configurator seems... extreme

I agree, but this isn't the case. `qmk_web_stack` not only runs the Configurator, but also the backend compiler, storage and web API it needs to talk to. It's almost entirely self contained; all it needs is to be able to download the `qmk_firmware` repo from GitHub.