qmk/qmk_configurator #1245: [Bug] shambles file cannot compile

ThePandaxXx, 2023-02-28:
### Describe the Bug

Heya there, 

Trying to flash this keyboard and when I compile the default keymap an error message keeps popping up saying

config.qmk.fm.says
Unexpected error while compiling

Not doing anything unique or different, the keyboard itself isn't plugged in, just compiling the keymap and the error message pops up.

keyboard-magpie, 2023-02-28:
This is not just shambles. Some issues with configurator after the breaking changes merge. Collaborator team are aware and working on it.

ThePandaxXx, 2023-02-28:
> This is not just shambles. Some issues with configurator after the breaking changes merge. Collaborator team are aware and working on it.

AH alright that makes more sense, thank you.

tzarc, 2023-02-28:
Try now.

closed 2023-03-06 by zvecr