qmk/qmk_configurator #1250: [Bug] Compile the keymap.json, which includes macros by qmk configurator; it doesn't work.

Jefferycheng, 2023-03-11:
### Describe the Bug

If I directly use the qmk firmware to compile, it works

keymap.json

```
{
  "version": 1,
  "notes": "",
  "documentation": "\"This file is a QMK Configurator export. You can import this at <https://config.qmk.fm>. It can also be used directly with QMK's source code.\n\nTo setup your QMK environment check out the tutorial: <https://docs.qmk.fm/#/newbs>\n\nYou can convert this file to a keymap.c using this command: `qmk json2c {keymap}`\n\nYou can compile this keymap using this command: `qmk compile {keymap}`\"\n",
  "keyboard": "xbows/nature",
  "keymap": "xbows_nature_layout",
  "macros": [
    [
      {
        "action": "tap",
        "keycodes": [
          "EQL"
        ]
      },
      {
        "action": "down",
        "keycodes": [
          "LSFT"
        ]
      },
      {
        "action": "tap",
        "keycodes": [
          "DOT"
        ]
      },
      {
        "action": "up",
        "keycodes": [
          "LSFT"
        ]
      }
    ]
  ],
  "layout": "LAYOUT",
  "layers": [
    [
      "QK_MACRO_0 "
    ]
  ]
}
```

yanfali, 2023-03-11:
I'm pretty sure this is an unsupported feature at this time. Configurator basically ignores the attribute for macros and doesn't pass it through.

Jefferycheng, 2023-03-13:
I will reopen the other issue as a feature request.

closed 2023-03-13 by Jefferycheng

yanfali, 2023-03-13:
Ya, at the very least, we can just pass through the data and let the API return the result.