qmk/qmk_configurator #1251: [Feature Request] Add the support of Macros

Jefferycheng, 2023-03-13:
### Description

Now, if I add the macros in the keymap.json, upload it to the configurator, and compile it, it doesn't work for my keyboard.

As #1250 mentioned, we wish to pass the macros object to compile as the first step.

And the next step, the web UI could set macros like https://keyboardlab.club.

Thanks a lot!

yanfali, 2023-03-13:
Wow, nice to see forks of qmk code with no attribution...

tzarc, 2023-03-13:
Yeah, and they're in breach of GPL by using QMK under the hood and not providing sources.

philvec, 2024-04-03:
Actually @Jefferycheng , the forked UI looks great and  the suyport for macros is very cool, good job - I would love to use it in my fork

How about switching the vibe  from 
### actually-acting-against-the-law
to 
### contributing-and-helping-other-geeks
? 💯

philvec, 2024-04-03:
can I get some feedback on why the comment was deleted? is this case going anywhere forward?

philvec, 2024-04-03:
Good point, the authors seem to have closed the business, the keybooards are no longer available
- but I found multiple YouTube review videos showing the unattributed fork and people commenting about how cool and useful it is - IMO this just shows spreading awareness of free software rights is more than needed...

In the topic of the feature request, I'm more than eager to help with the implementation, and I can start at once - the question is what do You mainteiners think about this particular approach?
 - defining macros as another key-value pair in `info.json`
 - defining macros similarily to keymaps or encoder_map, so sth like:
 ```
#if defined(MACRO_MAP_ENABLE)
enum macros {
    MC_0,
};

const uint16_t PROGMEM macro_map[][] = {
    [MC_0] =   { MC_TAP(KC_V), MC_DOWN(KC_S), MC_DLMS(100), MC_UP(KC_S) },
};
#endif
``` 

shall we? 👯‍♂️

philvec, 2024-04-03:
> Redundant, as its not a keyboard feature so should not change info.json and we already have

ofc, I meant `keymap.json` - can we skip the `keymap.c` compatibility then?

Then it is mainly about implementing the sheer functionality, using the defined macros - which should rather be a PR to `qmk_firmware` - if I'm correct, can You add it and make a reference here?

As for the UI, are You working on anything particular @Jefferycheng ? If not, then I will develop to our fork https://github.com/arrowmechanics/qmk_configurator and we'll see what we can get

zvecr, 2024-04-03:
> which should rather be a PR to qmk_firmware - if I'm correct, can You add it and make a reference here?

Theres no changes to make to the `qmk_firmware` repo. The functionality that needs to be implemented is purely client side and only isolated to this repo.