qmk/qmk_configurator #1252: [Bug] Old layout can't be loaded by configurator

KnorrFG, 2023-03-15:
### Describe the Bug

Hey, 
the following layout file, which was generated by the configurator doesn't load. After uploading it, the image of the keyboard disappears.

```json
{
  "version": 1,
  "notes": " ",
  "documentation": "\"This file is a QMK Configurator export. You can import this at <https://config.qmk.fm>. It can also be used directly with QMK's source code.\n\nTo setup your QMK environment check out the tutorial: <https://docs.qmk.fm/#/newbs>\n\nYou can convert this file to a keymap.c using this command: `qmk json2c {keymap}`\n\nYou can compile this keymap using this command: `qmk compile {keymap}`\"\n",
  "keyboard": "redox/rev1",
  "keymap": "redox_rev1_layout_mine",
  "layout": "LAYOUT",
  "layers": [
    [
      "KC_GRV",
      "KC_1",
      "KC_2",
      "KC_3",
      "KC_4",
      "KC_5",
      "KC_6",
      "KC_7",
      "KC_8",
      "KC_9",
      "KC_0",
      "KC_MINS",
      "KC_ESC",
      "KC_Q",
      "KC_W",
      "KC_E",
      "KC_R",
      "KC_T",
      "KC_MPLY",
      "KC_PSCR",
      "KC_Y",
      "KC_U",
      "KC_I",
      "KC_O",
      "KC_P",
      "KC_EQL",
      "KC_TAB",
      "KC_A",
      "KC_S",
      "KC_D",
      "KC_F",
      "KC_G",
      "KC_DEL",
      "ANY(KC_LEAD)",
      "KC_H",
      "KC_J",
      "KC_K",
      "KC_L",
      "KC_SCLN",
      "RCTL_T(KC_QUOT)",
      "KC_LSFT",
      "KC_Z",
      "KC_X",
      "KC_C",
      "KC_V",
      "KC_B",
      "LT(6,KC_MPRV)",
      "LT(3,KC_MNXT)",
      "KC_VOLD",
      "KC_VOLU",
      "KC_N",
      "KC_M",
      "KC_COMM",
      "KC_DOT",
      "ALL_T(KC_SLSH)",
      "KC_RSFT",
      "KC_NO",
      "ALL_T(KC_NO)",
      "MEH_T(KC_NO)",
      "LALT_T(KC_TILD)",
      "KC_LCTL",
      "MEH_T(KC_BSPC)",
      "KC_LGUI",
      "LT(2,KC_ENT)",
      "KC_SPC",
      "LT(1,KC_BSLS)",
      "KC_LEFT",
      "KC_DOWN",
      "KC_UP",
      "KC_RGHT"
    ],
    [
      "KC_TRNS",
      "KC_F1",
      "KC_F2",
      "KC_F3",
      "KC_F4",
      "KC_F5",
      "KC_F6",
      "KC_F7",
      "KC_F8",
      "KC_F9",
      "KC_F10",
      "KC_NLCK",
      "KC_TRNS",
      "KC_EXLM",
      "KC_AT",
      "KC_LCBR",
      "KC_RCBR",
      "KC_TILD",
      "KC_F11",
      "KC_F12",
      "KC_PSLS",
      "KC_P7",
      "KC_P8",
      "KC_P9",
      "KC_PMNS",
      "KC_PAUS",
      "KC_TRNS",
      "KC_HASH",
      "KC_DLR",
      "KC_LBRC",
      "KC_RBRC",
      "KC_PIPE",
      "KC_TRNS",
      "KC_TRNS",
      "KC_PAST",
      "KC_P4",
      "KC_P5",
      "KC_P6",
      "KC_PPLS",
      "KC_SLCK",
      "KC_TRNS",
      "KC_PERC",
      "KC_CIRC",
      "KC_LPRN",
      "KC_RPRN",
      "KC_BSLS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_UNDS",
      "KC_P1",
      "KC_P2",
      "KC_P3",
      "KC_PENT",
      "KC_NO",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_P0",
      "KC_P0",
      "KC_PDOT",
      "KC_PENT",
      "KC_NO"
    ],
    [
      "KC_NO",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_NO",
      "KC_NO",
      "LALT(KC_F4)",
      "LCTL(KC_RGHT)",
      "KC_END",
      "KC_NO",
      "KC_NO",
      "KC_TRNS",
      "KC_TRNS",
      "KC_NO",
      "KC_PGUP",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_HOME",
      "KC_NO",
      "KC_PGDN",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_TRNS",
      "KC_LEFT",
      "KC_DOWN",
      "KC_UP",
      "KC_RGHT",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "LCTL(KC_LEFT)",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO"
    ],
    [
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "RESET",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "RGB_TOG",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO"
    ],
    [
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "RESET",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO"
    ],
    [
      "KC_GRV",
      "KC_1",
      "KC_2",
      "KC_3",
      "KC_4",
      "KC_5",
      "KC_6",
      "KC_7",
      "KC_8",
      "KC_9",
      "KC_0",
      "KC_MINS",
      "KC_TAB",
      "KC_Q",
      "KC_W",
      "KC_E",
      "KC_R",
      "KC_T",
      "KC_NO",
      "KC_BSPC",
      "KC_Y",
      "KC_U",
      "KC_I",
      "KC_O",
      "KC_P",
      "KC_EQL",
      "KC_CAPS",
      "KC_A",
      "KC_S",
      "KC_D",
      "KC_F",
      "KC_G",
      "KC_ESC",
      "ANY(KC_LEAD)",
      "KC_H",
      "KC_J",
      "KC_K",
      "KC_L",
      "KC_SCLN",
      "KC_QUOT",
      "KC_LSFT",
      "KC_Z",
      "KC_X",
      "KC_C",
      "KC_V",
      "KC_B",
      "KC_NO",
      "CL_SWAP",
      "KC_NO",
      "KC_NO",
      "KC_N",
      "KC_M",
      "KC_COMM",
      "KC_DOT",
      "KC_SLSH",
      "KC_RSFT",
      "KC_LCTL",
      "KC_NO",
      "KC_LALT",
      "KC_LALT",
      "KC_SPC",
      "KC_SPC",
      "LT(6,KC_NO)",
      "KC_ENT",
      "KC_SPC",
      "KC_NO",
      "KC_LEFT",
      "KC_DOWN",
      "KC_UP",
      "KC_RGHT"
    ],
    [
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_F1",
      "KC_F2",
      "KC_F3",
      "KC_F4",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_F5",
      "KC_F6",
      "KC_F7",
      "KC_F8",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_F9",
      "KC_F10",
      "KC_F11",
      "KC_F12",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_NO"
    ]
  ],
  "author": "Anonym"
}

```

fauxpark, 2023-03-15:
Keyboard name should be `redox/rev1/base`.

KnorrFG, 2023-03-15:
ah, now it works again. Thanks

closed 2023-03-15 by KnorrFG