qmk/qmk_configurator #1260: [Bug] Keycodes for Alt and OS are swapped

blacktrub, 2023-05-28:
### Describe the Bug

Hello, I flash my keyboard with [bastardkb/charybdis/4x6/blackpill](https://github.com/qmk/qmk_firmware/tree/master/keyboards/bastardkb/charybdis/4x6/blackpill) firmware, but have a problem that my keycodes are swapped, for example KC_LALT and KC_LGUI, but the biggest problem is the KC_LGUI doesn't work at all. 

I have tried to to reset my EEPROM as suggested in [this issue](https://github.com/qmk/qmk_configurator/issues/479) and tried to press CG_NORM, but it didn't help.

Adding GUI_ON and press it helped me to bring my KC_LGUI button back alive, but I have the same problem when I unplug my keyboard.

I have tried to flash it with Linux (qmk tool) and Mac OS (qmk toolbox), but nothing changed.

yanfali, 2023-05-28:
Have you tried multiple computers?

blacktrub, 2023-05-28:
Yes, I have tried Mac OS laptop and Linux PC

Could it be OS specific problem?

yanfali, 2023-05-28:
If it moves with the keyboard, that's indicative of the problem being the
keyboard. There should be a way to clear the eeprom, or someone built the
firmware with strange defaults.

blacktrub, 2023-05-28:
I found how to print EEPROM config and there is what I found there
```
Bastard Keyboards:Charybdis (4x6) Blackpill:1: eeconfig:
Bastard Keyboards:Charybdis (4x6) Blackpill:1: default_layer: 255
Bastard Keyboards:Charybdis (4x6) Blackpill:1: debug_config.raw: FF
Bastard Keyboards:Charybdis (4x6) Blackpill:1: .enable: 1
Bastard Keyboards:Charybdis (4x6) Blackpill:1: .matrix: 1
Bastard Keyboards:Charybdis (4x6) Blackpill:1: .keyboard: 1
Bastard Keyboards:Charybdis (4x6) Blackpill:1: .mouse: 1
Bastard Keyboards:Charybdis (4x6) Blackpill:1: keymap_config.raw: FCFF
Bastard Keyboards:Charybdis (4x6) Blackpill:1: .swap_control_capslock: 1
Bastard Keyboards:Charybdis (4x6) Blackpill:1: .capslock_to_control: 1
Bastard Keyboards:Charybdis (4x6) Blackpill:1: .swap_lctl_lgui: 0
Bastard Keyboards:Charybdis (4x6) Blackpill:1: .swap_rctl_rgui: 0
Bastard Keyboards:Charybdis (4x6) Blackpill:1: .swap_lalt_lgui: 1
Bastard Keyboards:Charybdis (4x6) Blackpill:1: .swap_ralt_rgui: 1
Bastard Keyboards:Charybdis (4x6) Blackpill:1: .no_gui: 1
Bastard Keyboards:Charybdis (4x6) Blackpill:1: .swap_grave_esc: 1
Bastard Keyboards:Charybdis (4x6) Blackpill:1: .swap_backslash_backspace: 1
Bastard Keyboards:Charybdis (4x6) Blackpill:1: .nkro: 1
Bastard Keyboards:Charybdis (4x6) Blackpill:1: .swap_escape_capslock: 1
```

I have tried to reset EEPROM, but the output stays the same

Any idea how could I change it permanently? I mean this one `.no_gui: 1`

blacktrub, 2023-05-29:
I have solved my problem by changing `EEPROM_DRIVER` to value `vendor` in rules.mk

It seems like the problem was the wrong type of EEPROM driver.

closed 2023-05-29 by blacktrub