qmk/qmk_configurator #1290: [Feature Request] Converter/Convert To drop down for `pin_compatible` defined boards

dunk2k, 2023-07-12:
### Description

I do recall @zvecr mentioning something such as this in a qmk/qmk_firmware PR regarding reworking converters.

**Front End**:
If a keyboard in qmk_firmware repo has pin compatible defined, in `info.json` as `pin_compatible` or `development_board`, then a converter/"Convert To" drop down field appears in QMK configurator web front end near the _Compile_ and _Load Default_ buttons. 

Converter drop down would contain entries respective to keyboard's pin compatible definition. Drop down entries would include a default "Do Not Convert".

Hovering cursor over the drop down field's title would show a tooltip with words to effect of _convert this keyboard to a pin compatible controller_.

**Back End**:
If a converter drop down entry, that isn't "Do Not Convert", has been selected in web front end and Compile function begins, append `make` command with `CONVERT_TO=`_[respective user selected converter]_

dunk2k, 2023-07-12:
Would this be dependent on the QMK API accommodating QMK converters in the '/v1/compile' POST method?

daysgobye, 2023-10-08:
I want this and am happy to do the work I am 100% capable of doing it. If I can get some buy in / approval that this would be accepted. And if the core team has a way they would like to see this implemented please let me know, so I can do it right the first time.

mobergmann, 2024-05-25:
> I want this and am happy to do the work I am 100% capable of doing it. If I can get some buy in / approval that this would be accepted. And if the core team has a way they would like to see this implemented please let me know, so I can do it right the first time.

mobergmann, 2024-05-25:
> I want this and am happy to do the work I am 100% capable of doing it. If I can get some buy in / approval that this would be accepted. And if the core team has a way they would like to see this implemented please let me know, so I can do it right the first time.

I would also love to see this feature. Are there any news for this? I couldn't find any branches/MR.

dunk2k, 2024-05-25:
@daysgobye 
Would you still be willing to implement this feature?

Suffice to say both myself and @mobergmann are bought into this.