qmk/qmk_configurator #1291: [Bug]  Planck Rev7 fails to compile in web configurator

ThreepE0, 2023-07-14:
### Describe the Bug

Web configurator fails with error:

```
./.build/obj_planck_rev7_layout_ortho_4x12_mine/src/keymap.c:1,
                 from quantum/keymap_introspection.c:5:
./.build/obj_planck_rev7_layout_ortho_4x12_mine/src/keymap.c:17:253: error: 'EXT_PLV' undeclared here (not in a function)
```

Removing "EXT_PLV'" from the keymap results in a successful compile.

I'm not sure if the appropriate fix is to have the Planck firmware and default keymap updated, or to have the web configurator updated to support the keycode that is throwing the error here.

### Additional Context?

Using a fresh pull of the qmk firmware, and using a fresh copy of QMK_MSYS, version 1.7.2, the firmware compiles with no issue.

noroadsleft, 2023-07-15:
Apologies; I messed this up when I built the Configurator keymap for Planck rev7. This should be working shortly.

closed 2023-07-15 by noroadsleft

ThreepE0, 2023-07-15:
Confirmed working in the web configurator.  Thank you very much I appreciate it!

PR #1293 (2023-07-15): Fix planck/rev7 Default Keymap
## Description

[title]

Fixes #1291.