qmk/qmk_configurator #1297: [Feature Request] LED configuration

flp-CS, 2023-08-11:
### Description

The ability to configure the LED colours and behaviour would be a nice feature to have in the online configurator.

For example:
capslock colour change to indicate enable/disable.
single colour theme without crazy effects, where you can only change the brightness.
ability to enable/disable LEDs, like the side lights on gmmk models.

Or at least some single colour themes without any crazy rgb effects.

flpcstrike, 2024-12-19:
Keycodes in the keyboard settings tab needs to be updated, as I just noticed that "RGB_MODE_PLAIN", "RBG_HUD" and "RBG_HUI" are now deprecated. 

It would also be nice if the "RGB_MODE_PLAIN" keycode was restored or updated, as I can't seem to find the new keycode for this toggle, and having to cycle through every single rbg animation to get to plain single color mode seems a bit odd.

flpcstrike, 2024-12-19:
While I can understand the decision to not keep the keycodes for every single rgb effect, I can't understand the decision to remove the keycode that puts the LEDs into a static mode without any kind of rgb effect enabled.

> RGB_HUD/RGB_HUI have been renamed to UG_HUED/UG_HUEU (for RGBLight), or RM_HUED/RM_HUEU (for RGB Matrix).

I did notice that when looking up the keycodes when neither my old layout or online configurator codes worked as they are both deprecated, however, when changing the old keycodes into anykey and manually adding the new keycodes they initially didn't work.

Managed to fix it now by using `UG_PREV`/`UG_NEXT`, `UG_SATD`/ `UG_SATU` and `UG_HUED`/`UG_HUEU`, but it is a bit annoying having to toggle between every single rgb effect, hue and saturation. The ability to force a single rgb mode like a static color and rgb value/hex in the online configurator would be very nice to have.

fauxpark, 2024-12-19:
For both RGBLight and RGB Matrix, the "static colour" effect is always enabled and cannot be disabled. If a board has no effects enabled, and the internal config is somehow pointing at some other effect, then that is an invalid state; hitting the prev/next keycodes *should* fix it.

There have never been any keycodes to enforce a particular colour, only a particular effect.

I agree the keycode names here on the Configurator need to be updated.

flpcstrike, 2024-12-20:
>For both RGBLight and RGB Matrix, the "static colour" effect is always enabled and cannot be disabled.

Yes, but `RGB_MODE_PLAIN` doesn't work anymore, resulting in having to cycle through every rgb effect to find plain single color mode.

>There have never been any keycodes to enforce a particular colour, only a particular effect.

As the initial issue is a feature request about LED control in the online configurator, I thought it was obvious that I was talking about the ability to force an rgb mode and rgb value/hex in the online configurator, as well as me mentioning that it would be a nice feature to have specifically in the online configurator.

Akallabeth13666, 2025-05-22:
+1 it would be awsome to have