qmk/qmk_configurator #1303: qmk configurator doesn't give a .bin or .hex file

CrotoneFC, 2023-09-05:
### Information

- [X] None of the other issue templates apply

### Description

when I try to download my keymap, after having it compiled, the only file I can download is a .uf2, which I can't use and can't convert to anything useful. What do I do?

noroadsleft, 2023-09-06:
What keyboard is this regarding? Saying you only get a .UF2 and not a .BIN or .HEX doesn't tell me anything actionable.

CrotoneFC, 2023-09-06:
> What keyboard is this regarding? Saying you only get a .UF2 and not a .BIN or .HEX doesn't tell me anything actionable.

a tofu jr

noroadsleft, 2023-09-06:
If it's actually a Tofu Jr. PCB, .UF2 is the correct file format. It's an RP2040, which should show up like a USB flash drive when you enter Bootloader mode.

- [Flashing Instructions and Bootloader Information: Raspberry Pi RP2040 UF2](https://docs.qmk.fm/#/flashing?id=raspberry-pi-rp2040-uf2)

If it's a different bootloader, then your PCB isn't actually a Tofu Jr., and you'll need to set a different keyboard for your keymap.

CrotoneFC, 2023-09-07:
> If it's actually a Tofu Jr. PCB, .UF2 is the correct file format. It's an RP2040, which should show up like a USB flash drive when you enter Bootloader mode.
>
> * [Flashing Instructions and Bootloader [...]

thank you so much for the clarification, i couldn’t find any online just like this one. Do you by any chance know how can i check for sure if my pcb is an RP2040? i bought the entire kit from kbdfans so i think it should be the tofu jr pcb.

noroadsleft, 2023-09-07:
Did you buy the keyboard already assembled? The best way to tell - though not the easiest as far as effort - would be to take it apart and look at the microcontroller.

Have you put your keyboard into bootloader yet? Try unplugging the USB cable, holding the top left key, connecting the cable and then releasing the key. If a new storage drive appears on your PC, that should be the RP2040 bootloader. 

There's a short [build video for a Tofu Jr on YouTube here](https://youtu.be/zsm1dD_Jz2g?t=254 "Tofu Jr Keyboard Build using Drop x Invyr Holy Pandas - YouTube"). At 4:14 there's a decent shot of the PCB - one of the black squares on the bottom side should look like the image of an RP2040 microcontroller [as shown on raspberrypi.com](https://www.raspberrypi.com/documentation/microcontrollers/rp2040.html), if indeed your keyboard is a Tofu Jr.

CrotoneFC, 2023-09-07:
ok perfect. i managed to get to the bootloader mode and flash the new keymap but there's a problem. Some time ago I tried to remap the keyboard and I managed to do it on QMK configurator just by compiling it, and now the keys i modified that time won't change binding (e.g. i set the page up key to be an F10, and even if in the new keymap it is a page up again it still works as an F10)

CrotoneFC, 2023-09-07:
I'm pretty confused by the situation and very sorry for bothering yo so much, but you've been already a saviour to me, thank you so much

noroadsleft, 2023-09-07:
Curious: you're now the second person this week who's reported flashing a Configurator firmware to their board and not had the changes go into effect on their board. :raised_eyebrow:

CrotoneFC, 2023-09-07:
i mean it changed the keys that weren't already changed before, so that's puzzling to me

noroadsleft, 2023-09-07:
Do you have a JSON keymap export for your board? If so, could you post it here?

CrotoneFC, 2023-09-07:
{
  "version": 1,
  "notes": "",
  "documentation": "\"This file is a QMK Configurator export. You can import this at <https://config.qmk.fm>. It can also be used directly with QMK's source code.\n\nTo setup your QMK environment check out the tutorial: <https://docs.qmk.fm/#/newbs>\n\nYou can convert this file to a keymap.c using this command: `qmk json2c {keymap}`\n\nYou can compile this keymap using this command: `qmk compile {keymap}`\"\n",
  "keyboard": "dztech/tofu/jr/v1",
  "keymap": "dztech_tofu_jr_v1_layout_65_ansi_mine",
  "layout": "LAYOUT_65_ansi",
  "layers": [
    [
      "KC_ESC",
      "KC_1",
      "KC_2",
      "KC_3",
      "KC_4",
      "KC_5",
      "KC_6",
      "KC_7",
      "KC_8",
      "KC_9",
      "KC_0",
      "KC_MINS",
      "KC_EQL",
      "KC_BSPC",
      "KC_SLSH",
      "KC_TAB",
      "KC_Q",
      "KC_W",
      "KC_E",
      "KC_R",
      "KC_T",
      "KC_Y",
      "KC_U",
      "KC_I",
      "KC_O",
      "KC_P",
      "KC_LBRC",
      "KC_RBRC",
      "KC_BSLS",
      "KC_PGUP",
      "KC_CAPS",
      "KC_A",
      "KC_S",
      "KC_D",
      "KC_F",
      "KC_G",
      "KC_H",
      "KC_J",
      "KC_K",
      "KC_L",
      "KC_GRV",
      "KC_QUOT",
      "KC_ENT",
      "KC_PGDN",
      "KC_LSFT",
      "KC_Z",
      "KC_X",
      "KC_C",
      "KC_V",
      "KC_B",
      "KC_N",
      "KC_M",
      "KC_COMM",
      "KC_DOT",
      "KC_SCLN",
      "KC_RSFT",
      "KC_UP",
      "KC_END",
      "KC_LCTL",
      "KC_LGUI",
      "KC_LALT",
      "KC_SPC",
      "KC_RALT",
      "KC_RGUI",
      "KC_RCTL",
      "KC_LEFT",
      "KC_DOWN",
      "KC_RGHT"
    ],
    [
      "QK_GESC",
      "KC_F1",
      "KC_F2",
      "KC_F3",
      "KC_F4",
      "KC_F5",
      "KC_F6",
      "KC_F7",
      "KC_F8",
      "KC_F9",
      "KC_F10",
      "KC_F11",
      "KC_F12",
      "KC_DEL",
      "KC_HOME",
      "KC_TRNS",
      "RGB_TOG",
      "RGB_MOD",
      "RGB_HUI",
      "RGB_HUD",
      "RGB_SAI",
      "RGB_SAD",
      "RGB_VAI",
      "RGB_VAD",
      "KC_TRNS",
      "KC_PSCR",
      "KC_SCRL",
      "KC_PAUS",
      "QK_BOOT",
      "KC_PGUP",
      "KC_CAPS",
      "RGB_SPI",
      "RGB_SPD",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "EE_CLR",
      "KC_PGDN",
      "KC_LSFT",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "NK_TOGG",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_VOLU",
      "KC_MUTE",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_MPRV",
      "KC_VOLD",
      "KC_MNXT"
    ],
    [
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS"
    ],
    [
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS"
    ]
  ],
  "author": ""
}

CrotoneFC, 2023-09-07:
it wouldn't let me attach the json file so I took down the code from vscode, this would be the keymap I flashed

noroadsleft, 2023-09-07:
[image]
[image]

I don't see a way for you to even switch layers in your keymap.

You should perhaps go on [QMK Configurator's Key Tester](https://config.qmk.fm/#/test) and check that each key you press actually is doing what you expect it to do.

CrotoneFC, 2023-09-07:
i know that there's not a layer switch because I didn't put one, but I already checked on the key tester and by now the minus key works as an accent

noroadsleft, 2023-09-07:
What input language is your PC set to use?

CrotoneFC, 2023-09-07:
i'm really bugged it's like some key are hard wired to do a certain function and can't be changed

CrotoneFC, 2023-09-07:
> What input language is your PC set to use?

american english

noroadsleft, 2023-09-07:
> I already checked on the key tester and by now the minus key works as an accent

Has it worked like an accent the whole time you've had the board, or only since flashing new firmware? And is it always an accent, or do you sometimes get a minus sign?

CrotoneFC, 2023-09-07:
> > I already checked on the key tester and by now the minus key works as an accent
> 
> Has it worked like an accent the whole time you've had the board, or only since flashing new firmware? And is it always an accent, or do you sometimes get a minus sign?

no when I got the board it was default, but then I compiled the new bindings on qmk configurator and I don't know how it just worked like that. Now the keys that were changed always work like that, never got a minus or anything different

CrotoneFC, 2023-09-07:
maybe if there was a way to completely reset the keyboard and the flash in the new keymap it would work?

noroadsleft, 2023-09-07:
Try flashing the firmware in the attached ZIP file; I just compiled it.

It should be a standard US English QWERTY layout, except for the top two keys on the far-right column, which are `QK_BOOT` (Jump to Bootloader Mode) and `EE_CLR` (Clear EEPROM), respectively.

[image]

[dztech_tofu_jr_v1_crotonefc_test.zip](https://github.com/qmk/qmk_configurator/files/12553986/dztech_tofu_jr_v1_crotonefc_test.zip)

CrotoneFC, 2023-09-07:
everything works as it should except for the minus that still works as the accent

CrotoneFC, 2023-09-07:
again thank you so much for the effort you're putting in

noroadsleft, 2023-09-07:
Random thought; might be nothing: Tap the `EE_CLR` key that I placed, wait a bit, and then reboot the board (unplug then replug should be fine).

CrotoneFC, 2023-09-07:
nope, still the accent is there

CrotoneFC, 2023-09-07:
i can't really see a reason why it should be there still

CrotoneFC, 2023-09-07:
[image]

CrotoneFC, 2023-09-07:
yes

CrotoneFC, 2023-09-07:
rn i can't take a screen but i can confirm that the language has always been us english

CrotoneFC, 2023-09-07:
i'd leave it for the laptop kayboard which is italin layout

CrotoneFC, 2023-09-08:
i'm pretty sure it's not a problem of the language setting

CrotoneFC, 2023-09-08:
probably the question is senseless but is there a way to completely reset the PCB to factory settings and then flashing it afterwards?

yanfali, 2025-01-31:
This doesn't appear to be a configurator issue. Closing.

closed 2025-01-31 by yanfali