qmk/qmk_configurator #1304: [Bug] Compile Not Incorporating Changes

jackm767, 2023-09-06:
### Describe the Bug

I have loaded the below .json config (keyboard_layout.json), visually confirmed the layout on the GUI keyboard is correct, hit compile and downloaded the firmware (keyboard_layout.hex). However, when I got to flash my dz60 with the downloaded .hex file not all of the changes are incorporated into the layout actually on the board. I've attached both files and the json of the full source (full_source_map.json) below. Unsure what I can do to get the firmware to reflect the json. 

### Additional Context?

[json_and_hex.zip](https://github.com/qmk/qmk_configurator/files/12531649/json_and_hex.zip)

noroadsleft, 2023-09-06:
What specifically isn't updating from this compared to what happens on the board?

The .HEX file isn't particularly useful because it's only appropriate to flash on a DZ60 (which is not a board that I own).

jackm767, 2023-09-08:
Oops so sorry. User error with my local configs. Closing issue. Thanks for the response

closed 2023-09-08 by jackm767