qmk/qmk_configurator #1305: [Bug] crkdb_r2g not working with current .hex file

marcin-jakub-sroka, 2023-09-13:
### Describe the Bug

I think something is broken in compiling json layout for crkdb_r2g.

I changed my keyboard configuration in https://config.qmk.fm, compiled there to .hex file, downloaded both .hex and .json files, but after I flashed my keyboard with QMK Toolbox (https://github.com/qmk/qmk_toolbox/releases/tag/0.2.2) it **DOES NOT WORK properly**. Symptoms:
* At first glance you can see that something is wrong because the back light in right panel in not on
* Every few seconds it blinks in white (the left panel after flashing has default red back light)
* And the right pane either does not work (pressing keys has no effect) or (more often) every few seconds it generates random keystrokes (or may be not random - they might be just a few keystrokes pressed by me before)

First I thought that something was wrong with my changed layout (web configurator shouldn't generate wrong layout anyway) but I compiled old .json file and the result was pretty much the same. I tried upgrading QMK Toolbox with pre-release (https://github.com/qmk/qmk_toolbox/releases/tag/beta) with no success. I tried to compile locally my layout with docker image qmkfm/qmk_configurator:latest in Fedora WSL2 but the result is the same - .hex file is exactly the same and so the way my keyboard works after being flashed.

Last time I compiled my configuration on 26 of may.
Everything is OK after flashing with .hex file compiled on 26 of may.

yanfali, 2023-09-13:
Did you flash both sides? With enough time flashing just one side results in a broken keyboard.

marcin-jakub-sroka, 2023-09-13:
> Did you flash both sides? With enough time flashing just one side results in a broken keyboard.

What do you mean? The left side is attached to computer with USB cable, and the right to the left with TRRS cable.
I flash the same way with new .hex file, and with old .hex file. The process seams to work properly - there are no warnings or errors.
I use the following procedure:
* start QMK Toolbox
* disconnect keyboard
* connect while pressing Q key
* QMK Toolbox recognizes that keyboard is in flash-mode
* press [Clear EEPROM] button
* select .hex file
* press [Flash] button

marcin-jakub-sroka, 2023-09-13:
> Please also flash the right side of the keyboard with the same hex file and report back.

Ha! It's working!
Thanks guys!

Something must have changed since flashing separately was not necessary earlier.

So my procedure now is as follows:
* start QMK Toolbox
* disconnect keyboard
* disconnect keyboard parts (detach TRRS cable); not sure if this step is necessary, but it's what I did
* connect left part while pressing Q key
* QMK Toolbox recognizes that keyboard is in flash-mode
* press [Clear EEPROM] button
* select .hex file
* press [Flash] button
* disconnect keyboard part from computer
* repeat with right part of keyboard (use P key while attaching)
* attach parts with TRRS cable
* attach keyboard to computer

closed 2023-09-13 by marcin-jakub-sroka

marcin-jakub-sroka, 2023-09-13:
> For future reference:
> 
> https://docs.splitkb.com/hc/en-us/articles/360011949679-When-do-I-need-to-flash-my-microcontroller-
> 
> You can add to that list: if it has been a significant time since you last flashed.

Thanks. I'll remember that.
BTW Do you know that "sroka" in Polish is "magpie" in English?

yanfali, 2023-09-13:
One way to think about it is a split keyboard is two keyboards that
cooperate to act as a single keyboard.

If one keyboard has a newer version of the firmware it can't talk to the
other older because the protocol drifts over time as things are added or
removed.

If you keep that in mind it makes more sense to flash both sides if it's
been a while since you have done either side.