qmk/qmk_configurator #1323: [Bug] DC60 LAYOUT_60_ansi_arrow Backspace not working

josephchoi1116, 2023-11-17:
### Describe the Bug

I tried to flash 60 ansi arrow layout on my DC60 PCB, I found that the 2U backspace is not working.
By testing with a tweezer, I find that it has been configured as split backspace.

Is it a bug? Or that my pcb is faulty?

yanfali, 2023-11-18:
Isn't that the obviously correct configuration? 😂

Jokes aside, it's probably just how the key map was originally specified and the matrix is wrong in QMK. I may have access to this PCB and be able to troubleshoot it this weekend.

yanfali, 2023-11-18:
Sadly I have a clueboard PCB in my DC60. This is probably not an issue with configurator, but the original matrix on qmk_firmware. @noroadsleft can you take a look and see what you think?

yanfali, 2023-11-18:
This is a 5 year old PCB, and ALF is no longer in business. So finding people to test this is going to be a challenge.

josephchoi1116, 2023-11-18:
Thx everyone, I have found a workaround for this now.
Just use the default layout with split backspace, set both 1U to backspace, and it works with my 2U space now LOL

yanfali, 2023-11-18:
ah, so it's probably wired to one of them only. good work

noroadsleft, 2023-11-21:
> Is it a bug? Or that my pcb is faulty?

Bug. [I took a guess as to how the 2u Backspace was wired](https://github.com/qmk/qmk_firmware/blame/a6521b8521f56f39c3aa5054bf04307190de9035/keyboards/alf/dc60/matrix_diagram.md#L3-L6) when I added the additional layouts, because I don't own one of these. I set up the 2u Backspace as matrix position `[0, 14]`, but it's probably actually `[0, 13]`.

closed 2023-11-23 by zvecr