qmk/qmk_configurator #1332: [Bug] Key Mapping Rearranged Unexpectedly After JSON Import on kinesis/kintlc

conceptual-perception, 2024-02-25:
### Describe the Bug

I have a set of JSON files for a keyboard layout I've been modifying for the last two years. These json files have been used exclusively with QMK Configurator.

The keyboard is `kinesis/kintlc` My json is [ariadne-4-jan-2023-v2.json](https://github.com/qmk/qmk_configurator/files/14395376/ariadne-4-jan-2023-v2.json)

**Expected Behavior**
* Layer 0 looks like Dvorak

* Layer 3 looks like QWERTY

**Observed Behavior**

I import `ariadne-4-jan-2023-v2` and notice the following on layer 0

* Function keys on the home row
* Numerical keys where the function keys should be

There is other stuff wrong, but that's a start. 

### Additional Context?

[image]

tzarc, 2024-02-25:
I'd hazard a guess and say https://github.com/qmk/qmk_configurator/pull/1312 is the cause here.

tzarc, 2024-02-25:
Seems someone may have made some sort of conversion script on the kint side of things: https://github.com/kinx-project/kint/issues/76

yanfali, 2024-11-08:
ok, this seems resolved. so closing for now. thanks.

closed 2024-11-08 by yanfali